sudar/Arduino-Makefile #537: ATtiny instructions don't work, out of date

akkana, 2017-11-02:
First, thanks for this Makefile package. I use it for nearly all my Arduino development, and for most projects it's wonderfully flexible and easy to use.

ATtiny is a rare exception. There are two sets of conflicting instructions for using the Makefile with ATtiny: in Arduino.mk under ALTERNATIVE CORES, and in examples/ATtinyBlink. Neither set works with any of the cores I was able to find, so I think the instructions may be out of date, though it's possible that you have alternate sources that my googling wasn't able to find (in which case, links would be very appreciated).

**TLDR** if you'd prefer a patch suggesting updated instructions to those two files, rather than wade through all the long details below, let me know and I'll happily provide a patch and pull request.**/TLDR**

Most of the testing below is with the arduino-mk package from Debian testing (version 1.5.2-1) but I did check some of the recipes with the latest git (no difference), and I was using git master Arduino.mk for the instructions in all cases. Board is an ATtiny85 using a USBtinyISP as a programmer, on Debian testing.

Arduino.mk recommends "The HLT (attiny-master)" as the simplest. Since I had just gone through http://highlowtech.org/?p=1695, I took a guess "HLT" might stand for "highlowtech"; the JSON file they mention references https://github.com/damellis/attiny/archive/702aa287455f7e052cf94fd4949398fec0ef21b8.zip so I downloaded that and unzipped it into ~/sketchbook/hardware/attinyHLT/

Then I set up a Makefile following the Arduino.mk instructions, except using my programmer:
```
ALTERNATE_CORE = attinyHLT
BOARD_TAG = attiny85
ISP_PROG = usbtiny
include /usr/share/arduino/Arduino.mk
```
It failed:
avr-g++: error: missing device or architecture after ‘-mmcu=’

Inspection of boards.txt inside the core I'd downloaded suggested that I should try:
BOARD_TAG = attiny.menu.cpu.attiny85
but that gave lots of errors from the arduino core like:
```
/usr/share/arduino/hardware/arduino//cores/arduino/wiring.c: In function ‘__vector_5’:
/usr/share/arduino/hardware/arduino//cores/arduino/Arduino.h:76:45: error: expected expression before ‘/’ token
 #define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
                                             ^
```
It eventually turned out that I needed to define F_CPU. I found lots of old closed bugs here in the Arduino-Makefile project suggesting F_CPU = 16000000L and that fixed the compile, but then delays took 16x as long as they should. At least for attiny86, here's a Makefile that worked:
```
ALTERNATE_CORE = attinyHLT
BOARD_TAG = attiny.menu.cpu.attiny85
F_CPU = 1000000L
ISP_PROG = usbtiny
include /usr/share/arduino/Arduino.mk
```
I also tried with the newer version of that core: https://github.com/damellis/attiny/
which needed the same changes before it worked.

Arduino.mk also mentions a different core called arduino-tiny. I found that at https://code.google.com/archive/p/arduino-tiny/ and installed it to ~/sketchbook/hardware/arduino-tiny/
That one had some files in the wrong place. First, everything is under an avr directory, so you either need to reference it by that path, or mv avr/* . (I did the latter). Then, it has no boards.txt, only Prospective\ Boards.txt, and arduino-mk can't handle that so I copied that file to boards.txt. I got that working with this Makefile:
```
ALTERNATE_CORE = arduino-tiny
ISP_PROG = usbtiny
BOARD_TAG = attiny85at8
F_CPU  = 1000000L
include /usr/share/arduino/Arduino.mk
```
Like the damellis versions, it needed F_CPU defined, it needed the specific board tag, and it didn't work with the extra
```
ARDUINO_VAR_PATH  = ~/sketchbook/hardware/arduino-tiny/cores/tiny
ARDUINO_CORE_PATH = ~/sketchbook/hardware/arduino-tiny/cores/tiny
```
that Arduino.mk said it needed, so I had to remove those lines.

Just for completeness, I tried the third core mentioned in Arduino.mk: I found https://github.com/rexxar-tc/arduino-tiny.core2 and git cloned that into ~/sketchbook/hardware/arduino-tiny.core2. As with arduino-tiny, I had to cp Prospective\ Boards.txt boards.txt after which I tried:
```
ALTERNATE_CORE    = arduino-tiny.core2
BOARD_TAG         = attiny85at8
ISP_PROG          = usbtiny
ARDUINO_VAR_PATH  = ~/sketchbook/hardware/arduino-tiny.core2/cores/tiny
ARDUINO_CORE_PATH = ~/sketchbook/hardware/arduino-tiny.core2/cores/tiny
include /usr/share/arduino/Arduino.mk
```

but it failed with
cc1plus: fatal error: Arduino.h: No such file or directory
The file _~/sketchbook/hardware/arduino-tiny.core2/cores/tiny/Arduino.h_ does exist, so I guess ARDUINO_VAR_PATH and ARDUINO_CORE_PATH are enough to find include files there. avr-g++ was run with -I~/sketchbook/hardware/arduino-tiny.core2/cores/tiny but it looks like ~ doesn't work in -I options: if I set the Makefile variables to /home/username/sketchbook... I get past the Arduino.h error, but then it dies with
/home/akkana/sketchbook/hardware/arduino-tiny.core2/cores/tiny/tc_select_build_options.h:41:30: fatal error: tc_build_options.h: No such file or directory
 #include <tc_build_options.h>
and indeed there's no tc_build_options.h in ~/sketchbook/hardware/arduino-tiny.core2/cores/tiny. So maybe you should just drop that example unless you know a way to get it working.

Aside from the Arduino.mk comments, it would also be helpful to change examples/ATtinyBlink/Makefile to be one of the working examples, and maybe to add a link to where to get whichever core it actually uses. In particular "HLT" isn't clear at all, and I would never have found it except that I'd already run through the High-low tech IDE example.

I have the various Makefiles I tried, along with blink code, in https://github.com/akkana/arduino/tree/master/attiny85

sej7278, 2017-11-02:
all the attiny cores are a bit messed up with the 1.8 ide, but this works for me:

Makefile:
```make
ARDUINO_DIR = $(HOME)/arduino-1.8.5

#-------------- attinycore test:
#ALTERNATE_CORE = ATTinyCore
#BOARD_TAG = attinyx61
#BOARD_SUB = 861
#F_CPU = 8000000L

#-------------- arduino-tiny test:
ISP_PROG = usbasp
BOARD_TAG = attiny85at8
#BOARD_TAG = attiny2313at1
ALTERNATE_CORE = tiny

#-------------- damellis/hlt test:
#ISP_PROG = usbasp
#BOARD_TAG = attiny
#BOARD_SUB = attiny85
#ALTERNATE_CORE = attiny
#F_CPU = 16000000L

include /usr/share/arduino/Arduino.mk
```

and directory structure:
```
$ tree -L 2 ~/arduino16/hardware/
├── attiny
│   └── avr
├── ATTinyCore
│   ├── avr
│   ├── avrdude_conf.txt
│   ├── Installation.md
│   ├── Internals.md
│   ├── makefile.md
│   ├── README.md
│   └── Wiring.md
├── tiny
    ├── avr
    └── README
```

in fact the best place to look is in the [blink makefile](https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile#L78)

akkana, 2017-11-04:
Some progress: two out of four cores are working now, and some of my problem was due to using the arduino-mk package in Debian, some due to using arduino 1.8.4 instead of 1.8.5. But with both of those updated, another core that used to work no longer does.

Your attinycore test works if I use https://github.com/SpenceKonde/ATTinyCore. That was a core I hadn't found before.

For arduino-tiny, if I go to https://code.google.com/archive/p/arduino-tiny/downloads and download https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/arduino-tiny/arduino-tiny-0150-0020.zip and unzip it, there's an avr subdirectory but no README, so it doesn't correspond to what you have. If I use your arduino-tiny test, I get a long series of
grep: /home/akkana/sketchbook/hardware/tiny/avr/avr/boards.txt: No such file or directory
and indeed there's no boards.txt there, only 'Prospective Boards.txt'. If I cp 'Prospective Boards.txt'. boards.txt, though, it works. Where did you get your arduino-tiny that has boards.txt already in it?

(Searching some more for arduino-tiny also found https://github.com/Coding-Badly/arduino-tiny, which has a README but no avr subdirectory and no boards.txt either, so that probably isn't itl.)

Damellis still doesn't work. If I use https://github.com/damellis/attiny/ it doesn't have an avr directory inside it, just has
README.md  boards.txt  platform.txt  variants/
so it doesn't correspond to the damellis/hlt test in your example. Where does your damellis attiny come from? The older version of Arduino.mk didn't seem to need an avr directory but the current one apparently requires it. If I use your Makefile sample it complains
grep: /home/akkana/sketchbook/hardware/attiny/avr/boards.txt: No such file or directory
and then dies with
avr-g++: error: missing device or architecture after '-mmcu='
If I move everything into avr,
  cd ~/sketchbook/attiny; mkdir avr; mv * avr
then I don't see the "boards.txt: No such file" error but it still dies with "missing device or architecture." Funny, damellis is the one that worked with the fewest changes with arduino 1.8.4 and debian's Arduino.mk and now it doesn't work at all with 1.8.5 and git master Arduino.mk, so maybe it's set up for 1.8.4 and hasn't been updated yet.

arduino-tiny.core2 from https://github.com/rexxar-tc/arduino-tiny.core2 has the same problem as damellis: no avr subdirectory. Again, mkdir avr and moving everything into it just makes it break somewhere else.

So two out of the four are working (except for one needing a manual cp first). But it would definitely be useful to mention in the documentation where to get working versions of the cores, since the ones you have seem to be different from what I find using google or github, and some of the github versions don't work with Arduino.mk. Let me know if it would help if I made a documentation patch for the two that I got working (and warning against the two that don't).

I haven't gotten any of them to build with libraries yet (specifically Wire), but that's probably a symptom of the cores being messed up with 1.8, as you said -- it doesn't work with the IDE and the HLT core either.

Do you have any sense of which of these cores might be generally better? If I have to modify one of them to work with Wire I'm not sure which one to start with.

akkana, 2017-11-04:
(Ignore the question about Wire, I didn't realize Wire isn't compatible with ATtiny and that they need a special I2C library.)

sej7278, 2017-11-05:
a lot of those forks just haven't been updated to work with ide 1.5 or were abandoned before 1.6/1.8 came along. i tend to stick to ide 1.0.5 (debian package) and cores when using the attiny.

the Prospective Boards.txt file has to be edited and renamed to boards.txt - i believe it says that in the instructions for coding badly's [arduino-tiny](https://github.com/Coding-Badly/arduino-tiny) core, which is my favourite core (3 hardware pwm's) the 1.5+ version is [tinycore1](https://github.com/Coding-Badly/TinyCore1), not to be confused with the incomplete tinycore2.

the damellis/hlt [attiny](https://github.com/damellis/attiny) core seems very buggy (software 3x pwm simply doesn't work)

note most of the problems you're seeing are due to using the 1.8 ide and 1.0 cores/directories from what i can see, not the makefile.

sej7278, 2017-12-28:
More playing with cores, these all work with 1.8.5:

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.5
ISP_PROG       = usbasp
ALTERNATE_CORE = ATTinyCore
F_CPU          = 8000000L
BOARD_TAG      = attinyx4
BOARD_SUB      = 84
VARIANT        = tinyX4
ISP_LOW_FUSE   = 0xE2
include /usr/share/arduino/Arduino.mk
```

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.5
BOARD_TAG         = sanguino
BOARD_SUB         = atmega644p
ALTERNATE_CORE    = Sanguino
ISP_PROG     	  = usbasp
AVRDUDE_OPTS 	  = -v
include /usr/share/arduino/Arduino.mk
```

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.5
ALTERNATE_CORE = MightyCore
BOARD_TAG = 1284
MCU = atmega1284p
#BOARD_TAG = 644
#MCU = atmega644p
VARIANT = standard
F_CPU = 16000000L
include /usr/share/arduino/Arduino.mk
```

sej7278, 2017-12-28:
Fixed in PR #541

closed 2017-12-28 by sej7278