sudar/Arduino-Makefile #538: Is it possible to use my own main.cpp with my own main function with arduino libs?

GijsGoudzwaard, 2017-11-26:
Is it possible to use my own main.cpp with my own main function with arduino libs? I'm using CLion for arduino development and CLion doesn't support .ino files, so I'd like to use .cpp. I've got a main.cpp file and some other files. It works perfeclty when i'm using .ino but not with .cpp. 

This is my Makefile:
```BOARD_TAG                        = uno
MCU                              = atmega328p
F_CPU                            = 16000000

AVRDUDE_ARD_BAUDRATE           	 = 115200
MONITOR_BAUDRATE		 = 115200
ARDUINO_LIBS         		 = GraphicsLib MI0283QT9 SD digitalWriteFast SPI
LOCAL_CPP_SRCS			 = $(wildcard src/**/*.cpp)

-include Makefile.variable
include $(ARDMK_DIR)/Arduino.mk
```

I get this error:
```
/var/folders/mp/5kv9cvb13bd9qqpwswjhyrkr0000gn/T//ccMsihX3.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.startup+0x562): undefined reference to `setup'
<artificial>:(.text.startup+0x56a): undefined reference to `loop'
```

I've tried `NO_CORE_MAIN_CPP` but that doesn't work.

tuna-f1sh, 2017-11-28:
Have you defined the `setup` and `loop` functions in your main.cpp file? I believe you still have to do this and it's why the linker is complaining they don't exist.

GijsGoudzwaard, 2017-11-28:
By defining the `setup` and `loop` the same error persists.
I tried it like this:

```
void loop();
void setup();
```

And like this
```
void loop() {
}
void setup() {
}
```

But I keep getting the same errors. With a .ino file it works perfectly but I'd rather use a .cpp

tuna-f1sh, 2017-11-28:
I didn't realise you were redefining `main`. This works for me:

main.cpp
```
/* Project: ard-test */

#include <Arduino.h>
#include <Wire.h>

int main(void) {
  ;;
}
```
Makefile
```
BOARD_TAG = uno
TARGET = ard-test

LOCAL_CPP_SRCS = $(wildcard *.cpp)
NO_CORE_MAIN_CPP = true

include $(ARDMK_DIR)/Arduino.mk
```

Have done a `make clean`?

GijsGoudzwaard, 2017-11-28:
When using `NO_CORE_MAIN_CPP` I get the following error:
```
/usr/local/Cellar/avr-gcc/7.2.0/lib/gcc/avr/7.2.0/../../../../avr/lib/avr5/crtatmega328p.o:(.init9+0x0): undefined reference to `main'
collect2: error: ld returned 1 exit status
make: *** [build-uno/elite-combat.elf] Error```

tuna-f1sh, 2017-11-28:
So have you defined `main` in your file main.cpp? Like I say, the test I did above compiles with Arduino-Makefile so it should work. Can you share the source and Makefile? 

As I noted at the end of the last comment too, are you sure you want to re-define `main` if your main object is just to get your development environment to recognise the filetype? It might break lots of the Arduino functionality and libraries without the standard initialisation and serial polling.

GijsGoudzwaard, 2017-11-28:
I figured it out, I added my cpp files like so `LOCAL_CPP_SRCS = $(wildcard src/**/*.cpp)` which didn't include the main.cpp. Thanks for the help.

closed 2017-11-28 by GijsGoudzwaard