sudar/Arduino-Makefile #539: Is there an option that doesn't show all the make commands but does show the errors / avr memory usage?

GijsGoudzwaard, 2017-11-28:
Like the title says, I don't really care about all the commands it needs to run but I would like to see if there are warnings / errors and the avr memory usage.

tuna-f1sh, 2017-11-28:
It's a function built into `make`. From [man page](https://linux.die.net/man/1/make):

>-s, --silent, --quiet
Silent operation; do not print the commands as they are executed.

GijsGoudzwaard, 2017-11-28:
I tried that but it silences the whole output. Like I said I would like an option that doesn't show all the things it needs to compile and link and only the warnings, errors and avr memory usage.

tuna-f1sh, 2017-11-28:
It shouldn't silence the whole output; `make` will only compile what has changed by default so you will see nothing if you already have a current build. If you want to see something every time, do `make clean --silent && make --silent` - although this is not very efficient. I may not be what you are used to coming from the Arduino IDE as by default that rebuilds into a temporary location each time.

GijsGoudzwaard, 2017-11-28:
Thanks that was it. Would it be possible to make this a configurable option in the Makefile?

tuna-f1sh, 2017-11-28:
No not that I can think. It’s ‘make’ which invokes the Arduino Makefile configuration and what creates the command output stream. It’s up to the user to control that stream.

GijsGoudzwaard, 2017-11-29:
I fixed this issue by adding [MAKEFLAGS](http://www.gnu.org/software/make/manual/make.html#Options_002fRecursion) in my Makefile like so `MAKEFLAGS += --silent`

closed 2017-11-29 by GijsGoudzwaard

tuna-f1sh, 2017-11-29:
Ah ok, useful to know 👍 .