sudar/Arduino-Makefile #542: Eventual Python3 migration

sej7278, 2018-01-04:
```/usr/bin/env python``` still points to 2.7 for me on Debian Sid, so its far from a problem for now, but python2.7 is EOL in 2 years.

i'm just thinking ard-reset-arduino (and ardmk-init eventually) should probably change their shebangs to use ```/usr/bin/python``` instead - which already points to python3 for me.

Also the arduino-mk.spec and debian/control files will need updating to use python3-serial instead of python-serial (debian) and python3-pyserial instead of pyserial (fedora).

Need to check MacOS/cygwin implications and run by @maqifrnswa and @tuna-f1sh too.

sej7278, 2018-01-12:
@sudar any idea what's involved with macos homebrew to switch to python3?

Arduino.mk already has python3 support, the fedora/debian packaging just needs updating, and the python "bin" files, but i assume the documentation needs updating for homebrew at least?

i figure maybe we should do this during @tuna-f1sh's samd21 PR and give it a new version at the same time?

sudar, 2018-01-14:
@sej7278 

For homebrew, we don't explicitly include python as dependency.

I think with a little bit of documentation we can switch from Python 2 to Python 3.

sej7278, 2018-01-14:
Its more work that I thought, [here](https://github.com/sudar/Arduino-Makefile/compare/master...sej7278:python3?diff=split&name=python3) is a largely untested branch diffed against master

tuna-f1sh, 2018-01-15:
I think the biggest potential impact for this is with macOS users. It ships with an Apple compiled 2.7 but no 3. Most advanced users install homebrew Python 2 & 3 to avoid messing with the system Python as installing modules requires root - using /usr/bin/env currently ensures this version gets used rather than the system Python (some users might not have installed pyserial for the system Python). Additionally, /usr/bin/python3 will not resolve on a vanilla macOS install.

We would have to add a step of installing Python 3 to the homebrew instructions (until Apple adds it to their release). I believe homebrew creates a symbolic link to /usr/bin/python3.

sej7278, 2018-01-15:
well we've got over a year to look into it, but i thought we required homebrew anyway, not vanilla macos?

ladislas, 2018-01-15:
@sej7278 I'm not sure you need homebrew to use Arduino-Makefile as long as you have pyserial installed.

People can just clone the repo and use it from there after installing the Arduino app.

That being said, is requires a certain level of computer knowledge to start using the makefile and even know that you need it. I guess people at that point, using github and compiling from the command line, if they are on macOS, will all have homebrew installed. There not really any other way to dive into programming without it as most of the tools (if not all) are using it as some point.

tuna-f1sh, 2018-01-16:
@sej7278 homebrew doesn't install Python 3 by default so it would just need the lines adding:

```
brew install python3
pip3 install pyserial
```

sej7278, 2019-08-25:
i've rebased my python3 branch against master, is there anything more we need to add - this brew/pip3 stuff @tuna-f1sh ?

if the shebang is basically undoable on macos we'd have to replace all calls to bin files to ```python3 ard-reset-arduino``` which is awful, i guess we could check for the os first....

tuna-f1sh, 2019-08-26:
Yes, I use `/usr/bin/env python3` on macOS scripts now - that works.

ladislas, 2019-08-26:
@sej7278 @tuna-f1sh on macOS, `brew install python` now installs python 3 by default.

```console
→ brew info python
python: stable 3.7.4 (bottled), HEAD
Interpreted, interactive, object-oriented programming language
https://www.python.org/
/usr/local/Cellar/python/3.7.4 (3,967 files, 61.4MB) *
  Poured from bottle on 2019-08-14 at 12:10:39
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/python.rb
==> Dependencies
Build: pkg-config ✘
Required: gdbm ✔, openssl ✔, readline ✔, sqlite ✔, xz ✔
==> Options
--HEAD
	Install HEAD version
==> Caveats
Python has been installed as
  /usr/local/bin/python3

Unversioned symlinks `python`, `python-config`, `pip` etc. pointing to
`python3`, `python3-config`, `pip3` etc., respectively, have been installed into
  /usr/local/opt/python/libexec/bin

If you need Homebrew's Python 2.7 run
  brew install python@2

You can install Python packages with
  pip3 install <package>
They will install into the site-package directory
  /usr/local/lib/python3.7/site-packages

See: https://docs.brew.sh/Homebrew-and-Python

```

tuna-f1sh, 2019-08-26:
That's good, but it doesn't change the install steps much:
```
brew install python
pip3 install pyserial
```
And I think one still has to use `/usr/bin/env python3` to be cross compatible with all systems, otherwise this would need to be `/usr/local/bin/python3` on macOS unless I'm mistaken.

ladislas, 2019-08-26:
No you're right, it just makes it easier on macOS as people using brew don't have to think about it at all.

sej7278, 2019-08-26:
ok, so i updated my fork to use virtualenv (seems to work ok on debian) and added the homebrew instructions to README.md, although i'm not very happy with how clear they are - i've sort of repeated the pyserial instructions and added the brew installation of python3 to the macports bit!

not sure how we get this pushed to debian as @maqifrnswa seems to have dropped off the internet at the end of 2017, hope nothing awful has happened there :-(

sej7278, 2019-09-26:
did some experimentation and came to the same conclusion as #616 that fedora (and debian) hates virtualenv.

any shebangs in packages (rpm/deb) have to be absolute, so /usr/bin/python3 not /usr/bin/env python3

so not sure where that leaves us with macos again (symlink /usr/local/bin/python3 to /usr/bin/python3 ?)

i made a fedora 30 rpm using hardcoded python3 shebangs [here](https://www.the-jedi.co.uk/downloads/arduino/arduino-mk-1.6.0-2.fc30.noarch.rpm)

sej7278, 2019-11-12:
Probably need to get this moving: https://lists.debian.org/debian-devel-announce/2019/11/msg00000.html

@maqifrnswa ?

tuna-f1sh, 2019-11-13:
After looking into this a few times to support both macOS and Debian, the only solution appears to be to call `sh` as the shebang to detect the user python3 installation, then pass this to it. [1](https://blogs.gnome.org/mcatanzaro/2018/02/16/on-python-shebangs/) [2](https://dxr.mozilla.org/mozilla-central/rev/994a8d6eccbcdc6106794705bd77e3ac5f031be2/mach#1-10)

Please see: https://github.com/tuna-f1sh/Arduino-Makefile/commit/284542c76c0a8a81b5030ec06b955943dbf56b43

Works in macOS and my Arch installation. Can you test with Debian? It feels dirty but if it works...

sej7278, 2019-11-13:
fedora and debian won't like that - they have specific policies about versioned shebangs and shelling out from one language to another is dirty, bad tuna bad! ;-)

tuna-f1sh, 2019-11-14:
Ha, you're right, not sure what came over me.

Looking at my macOS machine, it appears Python 3 is now distrubuted with the base install (Catalina), so there is a `python3` in /usr/bin independant of the Homebrew install to /usr/local/bin.

This means we can use `/usr/bin/python3` in the scripts and change the macOS install instructions to (no Homebrew steps):

`sudo /usr/bin/python3 -m pip pyserial`

With the caveat that users must be using macOS 10.15 or have `python3` linked to /usr/bin. Seems like we have a solution.

sej7278, 2019-11-14:
i made a PR with @tuna-f1sh's new mac instructions if people could review/test please

https://github.com/sudar/Arduino-Makefile/pull/620

ladislas, 2019-11-14:
My $0.02: I would not recommend macOS users to use the base python install to install pyserial or anything else as it forces the user to use `sudo`.

macOS developers tend to always have (or at least should have) homebrew installed as it keeps the system clean in case something stops working. It's the same with gems and ruby.

So for macOS instructions I would say:

- Install Homebrew (http://brew.sh/)
- `brew install python`
- `pip3 install -U pyserial`

And voilà.

tuna-f1sh, 2019-11-14:
@ladislas I'm in complete agreement there regarding macOS and base Python but the problem we are facing is that Homebrew installs to /usr/local/bin and not /usr/bin ([see your previous comment](https://github.com/sudar/Arduino-Makefile/issues/542#issuecomment-524773308)), which if we are to remove the `/usr/bin/env python` means the scripts will no long work using Homebrew Python.

tuna-f1sh, 2019-11-14:
Maybe the better solution in all this is to actually call the scripts with `python3`; they can still install the shebangs to /usr/bin/ for standalone use but most users will not have to worry about this.

For example, pre-append `python3` here: https://github.com/sudar/Arduino-Makefile/blob/master/Arduino.mk#L860

ladislas, 2019-11-14:
`/usr/local/bin` is in the default macOS path, so `#!/usr/bin/env python3` will work.

sej7278, 2019-11-14:
we can't use virtualenv @ladislas due to linux packaging requirements

ladislas, 2019-11-14:
> we can't use virtualenv ladislas due to linux packaging requirements

I'm not a python expert, but I'm not using virtualenv on my machine. Is calling `#!/usr/bin/env python3` using virtualenv?

sej7278, 2019-11-14:
yes

ladislas, 2019-11-14:
Alright, my bad, I did not know. This [post](https://blogs.gnome.org/mcatanzaro/2018/02/16/on-python-shebangs/) made it clearer to me :)

Then I think @tuna-f1sh's idea is the best: https://github.com/sudar/Arduino-Makefile/issues/542#issuecomment-553853903

tuna-f1sh, 2019-12-21:
Finally got around to implementing https://github.com/sudar/Arduino-Makefile/issues/542#issuecomment-553853903

https://github.com/tuna-f1sh/Arduino-Makefile/commit/87d524107147af571abf2a1515e98f5422c5f844

sej7278, 2020-08-04:
Looks like we need to get a move on here, Debian Sid just tried to remove arduino-mk as its now removing python2.

So @tuna-f1sh do you have a PR with your `PYTHON_CMD` bits which hardcodes the shebang to `#!/usr/bin/python3` as well as my bits from #620 ? I think your commit doesn't cover all of the documentation changes for linux, ard-reset-arduino and the specfile?

tuna-f1sh, 2020-08-05:
@sej7278 merge into my fork and all works fine for me on macOS and Arch. Thanks for doing the grunt work.

There are some other changes relating to the SAMD support in my fork now but I should probably make a separate pull request.

Will be good when all this is rolled into a release so that other package managers can be updated too. Most of them are tracking something from 3 years ago!

sej7278, 2020-08-05:
@tuna-f1sh yeah, very keen on getting some of your SAM stuff merged and a couple of other PR's (examples and such) once we've done this python3 thing, then probably call it a 1.7 release or something.

tuna-f1sh, 2020-08-06:
> be nice to fix this fecking TravisCI too!

https://github.com/sudar/Arduino-Makefile/pull/640 🤙

sej7278, 2020-08-06:
@tuna-f1sh any chance you could pull in #630 too - its basically a clean up of the examples, then i think maybe we'll just merge your giant PR

tuna-f1sh, 2020-08-06:
@sej7278 it's in!

sej7278, 2020-08-06:
@sudar do you fancy merging the world's largest PR here? ;-)

sej7278, 2020-08-06:
oh could you note that it closes #616, #635 (python3) and #575 (examples) in HISTORY.md ?

Probably fixes #611 and #581 (samd) too doesn't it?

sudar, 2020-08-06:
@sej7278 Definitely! 

Just give me a day or two, I will test it a bit and will merge it.

@sej7278 Once this is merged, I am thinking of bumping up the version to v2.0.0, since this is a backward incompatible change. What are your thoughts?

Also thank you @sej7278 @tuna-f1sh for all the hard work in adding support for python3.

sej7278, 2020-08-07:
@sudar yeah might be a better idea than 1.7.0

closed 2020-09-16 by sej7278