sudar/Arduino-Makefile #544: lib-detection gets stuck searching for dependencies

rpbeltran, 2018-01-10:
I went to compile my code to find that it was never finishing (after running for about ~45 minutes). When I would interrupt the process it would display a traceback to `Arduino-Makefile/bin/lib-detection` somewhere around line 70 ( `currentFile = open(os.path.join(USER_LIB_PATH, lib, lib + ".h"))` )

For some reason, it was getting stuck calculating `SKETCH_LIBS_DEPS`. As a workaround, I specified every library and every library dependency directly in the `ARDUINO_LIBS` variable of the makefile, commented out that section of lib-detection, and replaced it with `SKETCH_LIBS_DEPS = []`. This approach seems to be working for now.

[image]

My project is somewhat large with 15 libraries specified in my USER_LIB_PATH. Each individual library is fairly simple, with a handful of includes and an average size of ~200 - 300 lines, with a couple libraries being substantially larger. Some libraries include other libraries. Hardcoding dependencies in the makefile without commenting out lib-detection did not help.

Unfortunately, I cannot upload my project as it's for work and parts of it are confidential, but would be happy to try and provide any specific information that might be helpful.

Thanks!

ladislas, 2018-01-10:
@rpbeltran are you using the Bare arduino project?

rpbeltran, 2018-01-10:
Yeah, you caught me just as I was editing my post, I'll move this there.

closed 2018-01-10 by rpbeltran