sudar/Arduino-Makefile #547: undefined reference to setup / loop

achu1431, 2018-01-24:
./opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/266865351/build -hardware opt/arduino-builder/hardware -hardware ./opt/cores -tools opt/arduino-builder/tools -tools ./opt/tools -built-in-libraries opt/libraries/latest -libraries /tmp/266865351/pinned -libraries /tmp/266865351/custom -fqbn arduino:avr:uno -build-cache /tmp -logger humantags -verbose=false /tmp/266865351/sketch_jan24a

/tmp/ccGoZPWp.ltrans0.ltrans.o: In function `main':

/home/admin/builder/opt/cores/arduino/avr/cores/arduino/main.cpp:43: undefined reference to `setup'

/home/admin/builder/opt/cores/arduino/avr/cores/arduino/main.cpp:46: undefined reference to `loop'

collect2: error: ld returned 1 exit status

exit status 1

achu1431, 2018-01-24:
could'nt find the solution for long time..pls help me frnds,,,tnx in advance

sudar, 2018-01-24:
@achu1431 

Can you post your full makefile and source file?

sej7278, 2018-01-24:
if he's calling arduino-builder he's not even using the makefile.

i seem to recall some old problem where you can't have your own main()