sudar/Arduino-Makefile #552: Clock speed on teensy31 defaults to 168000000L

gdhamp, 2018-02-01:
Because clock speeds higher than 96MHz are defined in teensy boards.txt (168MHz, 144MHz and 120MHz) and Teensy.mk uses the highest defined clock speed, 168Mhz is selected for F_CPU and this doesn't work. I tested by defining F_CPU in my Makefile and with F_CPU=96000000L the code uploaded and ran properly.

tuna-f1sh, 2018-02-01:
I haven't used Teensy so don't have experience with the different boards but Teensy.mk sets to 168 MHz since it chooses the last entry it multiple speeds are defined and no BOARD_SUB has been defined: https://github.com/sudar/Arduino-Makefile/blob/master/Teensy.mk#L47

If your board doesn't work at 168 MHz, is it a subcategory of board?

gdhamp, 2018-02-01:
I actually think it's a bit of an oversight by the Teensy people.
Boards.txt for the Teensy 3.1 has the configuration items for the Arduino
menu commented out for the higher clock speeds but keeps the config item
for the actual clock speed. I'll log an issue against that and see if Paul
will fix it. I'm OK with things right now but I'd hate to see people try
your awesome makefile and have it not work.

https://github.com/sudar/Arduino-Makefile/blob/master/Teensy.mk#L47