sudar/Arduino-Makefile #553: Another -mmcu missing parameter

raycewest, 2018-02-18:
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/share/arduino 
- [USER]               ARDMK_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 185 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /root/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /root/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE = sparkfun 
- [COMPUTED]           ALTERNATE_CORE_PATH = /root/Arduino/hardware/sparkfun/avr  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
- [COMPUTED]           ARDUINO_VAR_PATH = /root/Arduino/hardware/sparkfun/avr/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = /root/Arduino/hardware/sparkfun/avr/boards.txt (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /root/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 16MHzatmega328 
- [USER]               BOARD_TAG = promicro16 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-promicro16-16MHzatmega328 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 7.3.0 (avr-gcc)
-------------------------
mkdir -p build-promicro16-16MHzatmega328
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU= -DARDUINO=185 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/root/Arduino/hardware/sparkfun/avr/variants/    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto Blink.ino -o build-promicro16-16MHzatmega328/Blink.ino.o
avr-g++: error: missing device or architecture after '-mmcu='
make: *** [/usr/share/arduino/Arduino.mk:1256: build-promicro16-16MHzatmega328/Blink.ino.o] Error 1

boards.txt is correctly located.  Any help?

sej7278, 2018-02-18:
there isn't a sparkfun promicro that runs an atmega328 - they all run 32u4's, so your variant and core aren't being computed. this works for me:

```make
BOARD_TAG = promicro
BOARD_SUB = 16MHzatmega32U4
ALTERNATE_CORE = sparkfun

include /usr/share/arduino/Arduino.mk
```

also you really shouldn't be running it as root.

raycewest, 2018-02-18:
No change I'm afraid.

-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/share/arduino 
- [USER]               ARDMK_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 185 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /root/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /root/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE = sparkfun 
- [COMPUTED]           ALTERNATE_CORE_PATH = /root/Arduino/hardware/sparkfun/avr  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
- [COMPUTED]           ARDUINO_VAR_PATH = /root/Arduino/hardware/sparkfun/avr/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = /root/Arduino/hardware/sparkfun/avr/boards.txt (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /root/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 16MHzatmega32u4 
- [USER]               BOARD_TAG = promicro16 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-promicro16-16MHzatmega32u4 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino/avr/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 7.3.0 (avr-gcc)
-------------------------
mkdir -p build-promicro16-16MHzatmega32u4
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -mmcu= -DF_CPU= -DARDUINO=185 -DARDUINO_ARCH_AVR -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino/avr/cores/arduino -I/root/Arduino/hardware/sparkfun/avr/variants/    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto Blink.ino -o build-promicro16-16MHzatmega32u4/Blink.ino.o
avr-g++: error: missing device or architecture after '-mmcu='
make: *** [/usr/share/arduino/Arduino.mk:1256: build-promicro16-16MHzatmega32u4/Blink.ino.o] Error 1

raycewest, 2018-02-18:
Just saw your remark about root - I know I shouldn't be, but can't do it otherwise, despite chown'ing Arduino.mk and the Makefile to me.  Still trying to figure that out.