sudar/Arduino-Makefile #554: Arduino.mk:  prerequisites cannot be defined in recipes

raycewest, 2018-02-18:
"make -f ../../Makefile lets_split/rev2:default:avrdude" gives:
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /usr/share/arduino 
- [USER]               ARDMK_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 185 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/rayce/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/rayce/Arduino (from arduino preferences file)
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /usr/share/arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [USER]               ALTERNATE_CORE = sparkfun 
- [COMPUTED]           ALTERNATE_CORE_PATH = /home/rayce/Arduino/hardware/sparkfun/avr  (from ARDUINO_SKETCHBOOK and ALTERNATE_CORE)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/rayce/Arduino/hardware/sparkfun/avr/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = /home/rayce/Arduino/hardware/sparkfun/avr/boards.txt (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /home/rayce/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 16MHzatmega32U4 
- [USER]               BOARD_TAG = promicro 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = promicro (from build.variant)
- [COMPUTED]           OBJDIR = build-promicro-16MHzatmega32U4 (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = /usr/share/arduino/archlinux-arduino/avr/cores/arduino/ 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 -flto -fno-fat-lto-objects 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 -fno-threadsafe-statics -flto 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 7.3.0 (avr-gcc)
-------------------------
/usr/share/arduino/Arduino.mk:1185: *** prerequisites cannot be defined in recipes.  Stop.

I'm trying to integrate [ADNS-9800 code](https://github.com/mrjohnk/ADNS-9800/tree/master/Arduino%20Example%20Sketches) into a [QMK keyboard firmware](https://github.com/qmk/qmk_firmware/tree/master/keyboards/lets_split) in order to integrate a trackball into the keyboard via the same Pro Micro - trying to use Arduino-Makefile as the IDE abstracts away a lot of the needed includes.  Thanks for any pointers!

sej7278, 2018-02-18:
what's `lets_split/rev2:default:avrdude` about?

have you tried just calling make from the same directory as your Makefile?

raycewest, 2018-02-18:
Yeah I was trying to mix - new to Arduino, thought it might be possible. I added your typical makefile lines and sourced Arduino.mk to that existing (lets_split, which yes, doesn't use Arduino) one.