sudar/Arduino-Makefile #555: unrecognized option '-plugin'

jmbldwn, 2018-03-11:
I'm using this tool to try and bring all dependencies into my repository to reduce dependencies others have on their environments.  

I have everything compiling at this point, but it's breaking in the linker with this message:
```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -mmcu=atmega32u4 -Wl,--gc-sections -Os -flto -fuse-linker-plugin -o build-feather32u4/am.elf build-feather32u4/device.ino.o build-feather32u4/libcore.a   -lc -lm
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: unrecognized option '-plugin'
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: use the --help option for usage information
collect2: error: ld returned 1 exit status
make: *** [build-feather32u4/am.elf] Error 1
```
It looks like its unhappy with the -fuse-linker-plugin argument, which is added from Arduino.mk here:
```
ifeq ($(shell expr $(CC_VERNUM) '>' 490), 1)
    LDFLAGS += -flto -fuse-linker-plugin
endif
``` 
I've pulled all of the libraries I depend on into a local 'libraries' folder, and also copied out the arduino 'packages' directory locally.

Here's my Makefile as it stands:
```
ARDUINO_SKETCHBOOK = .
#ARDUINO_DIR=arduino
ARDUINO_CORE_PATH=packages/arduino/hardware/avr/1.6.21/cores/arduino
ALTERNATE_CORE_PATH=packages/adafruit/hardware/avr/1.4.12
ARDUINO_LIBS = Ethernet2 SPI dimmer Adafruit_Motor_Shield_V2_Library Wire
ARCHITECTURE = avr
BOARDS_TXT=boards.txt
BOARD_TAG=feather32u4
include /usr/local/opt/arduino-mk/Arduino.mk
```

jmbldwn, 2018-03-11:
Duh.  Didn't take long to notice I hadn't explicitly set AVR_TOOLS_DIR and it was using the IDE's version instead.

```AVR_TOOLS_DIR=/usr/local```

closed 2018-03-11 by jmbldwn

sej7278, 2018-03-12:
avr-gcc 4.9.2 that was bundled with the ide should have coped with the lto stuff though, what version ide was it? a bit of googling suggests its an osx bug with ld, probably fixed by using a newer ide

jmbldwn, 2018-03-12:
It's the most recent IDE on the mac, 1.8.5.

sej7278, 2018-03-12:
does the ide build your sketch with LTO enabled?

jmbldwn, 2018-03-29:
Good question.  One thing I've noticed is the output of the makefile build is noticeably smaller.  I'm not sure how to figure out what the IDE is doing.