sudar/Arduino-Makefile #557: The diagnostics-color compiler switch should be modifiable.

wingunder, 2018-03-18:
Some IDEs do not work nicely with the recently added diagnostics-color compiler switch. PR #556 adds the DIAGNOSTICS_COLOR_WHEN variable, for controlling diagnostics-color switch. The default was chosen to be 'always', which makes the Arduino.mk file backwards compatible with everything up to commit fa82c3a9db (Sat Sep 2 15:32:52 2017 +0100).
    
The supported values for DIAGNOSTICS_COLOR_WHEN are: `always`, `never` and `auto`, and as with all Arduino.mk variables, it can easily be overridden. For more details on the diagnostics-color compiler switch, see: https://gcc.gnu.org/onlinedocs/gcc-4.9.2/gcc/Language-Independent-Options.html#Language-Independent-Options

sej7278, 2018-03-18:
merged thanks

closed 2018-03-18 by sej7278

wingunder, 2018-03-18:
Great, thanks! :)