sudar/Arduino-Makefile #560: Calling <<-EOS.undent is disabled!

j4zzcat, 2018-04-16:
I'm getting this during install.
```
Error: Calling <<-EOS.undent is disabled!                       
Use <<~EOS instead.             
/usr/local/Homebrew/Library/Taps/sudar/homebrew-arduino-mk/arduino-mk.rb:22:in `caveats'                                        
Please report this to the sudar/arduino-mk tap!                 
Or, even better, submit a PR to fix it!  
```

tuna-f1sh, 2018-04-16:
This is a problem with the Homebrew formula, not Arudino-Makefile so you should make the issue there: https://github.com/sudar/homebrew-arduino-mk 

If you want a quick fix, it looks like you could try changing [line 22](https://github.com/sudar/homebrew-arduino-mk/blob/master/arduino-mk.rb#L16) to <<~EOS as it suggests.

j4zzcat, 2018-04-16:
Done.

closed 2018-04-16 by j4zzcat