sudar/Arduino-Makefile #562: recipe for target 'upload' failed

zagatta-sonah, 2018-04-25:
I want to use the Makefile to push code to the [SleepyPi](https://spellfoundry.com/product/sleepy-pi-2/)

My Makefile looks linke this:
```
### DISCLAIMER
### This is an example Makefile and it MUST be configured to suit your needs.
### For detailed explanations about all of the available options, please refer
### to https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md
### Original project where this Makefile comes from: https://github.com/WeAreLeka/Bare-Arduino-Project

### PROJECT_DIR
### This is the path to where you have created/cloned your project
PROJECT_DIR       = /sonah/src/sleepypi/run/script2

### ARDMK_DIR
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = $(PROJECT_DIR)

### ARDUINO_DIR
### Path to the Arduino application and resources directory.
ARDUINO_DIR       = /usr/share/arduino

### USER_LIB_PATH
### Path to where the your project's libraries are stored.
USER_LIB_PATH    :=  /usr/share/arduino/libraries

### BOARD_TAG
### It must be set to the board you are currently using. (i.e uno, mega2560, etc.)
BOARD_TAG         = sleepypi

### MONITOR_BAUDRATE
### It must be set to Serial baudrate value you are using.
MONITOR_BAUDRATE  = 115200

### AVR_TOOLS_DIR
### Path to the AVR tools directory such as avr-gcc, avr-g++, etc.
AVR_TOOLS_DIR     = /usr

### AVRDUDE
### Path to avrdude directory.
AVRDUDE          = /usr/bin/avrdude

### CFLAGS_STD
### Set the C standard to be used during compilation. Documentation (https://github.com/WeAreLeka/Arduino-Makefile/blob/std-flags/arduino-mk-vars.md#cflags_std)
CFLAGS_STD        = -std=gnu11

### CXXFLAGS_STD
### Set the C++ standard to be used during compilation. Documentation (https://github.com/WeAreLeka/Arduino-Makefile/blob/std-flags/arduino-mk-vars.md#cxxflags_std)
CXXFLAGS_STD      = -std=gnu++11

### CXXFLAGS
### Flags you might want to set for debugging purpose. Comment to stop.
CXXFLAGS         += -pedantic -Wall -Wextra

### MONITOR_PORT
### The port your board is connected to. Using an '*' tries all the ports and finds the right one.
MONITOR_PORT      = /dev/ttyS0

### CURRENT_DIR
### Do not touch - used for binaries path
CURRENT_DIR       = $(shell basename $(CURDIR))

### OBJDIR
### This is where you put the binaries you just compile using 'make'
OBJDIR            = $(PROJECT_DIR)/bin/$(BOARD_TAG)/$(CURRENT_DIR)

### Do not touch - the path to Arduino.mk, inside the ARDMK_DIR
include $(ARDMK_DIR)/Arduino.mk
```

And when I run it i get the following message:
```
root@raspberrypi3-c49223e:/sonah/src/sleepypi/run/script2# make upload
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX
- [USER]               ARDUINO_DIR = /usr/share/arduino
- [USER]               ARDMK_DIR = /sonah/src/sleepypi/run/script2
- [AUTODETECTED]       ARDUINO_VERSION = 105
- [DEFAULT]            ARCHITECTURE =
- [DEFAULT]            ARDMK_VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK =
- [USER]               AVR_TOOLS_DIR = /usr
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = /usr/share/arduino/libraries
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = sleepypi
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [USER]               OBJDIR = /sonah/src/sleepypi/run/script2/bin/sleepypi/script2
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
-  [USER]              MONITOR_BAUDRATE = 115200
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [USER]               CFLAGS_STD = -std=gnu11
- [USER]               CXXFLAGS_STD = -std=gnu++11
- [COMPUTED]           DEVICE_PATH = /dev/ttyS0 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6
- [COMPUTED]           CC_VERSION = 4.8.1 (avr-gcc)
-------------------------
mkdir -p /sonah/src/sleepypi/run/script2/bin/sleepypi/script2
make reset
make[1]: Entering directory '/sonah/src/sleepypi/run/script2'
/sonah/src/sleepypi/run/script2/bin/ard-reset-arduino  /dev/ttyS0
make[1]: /sonah/src/sleepypi/run/script2/bin/ard-reset-arduino: Command not found
/sonah/src/sleepypi/run/script2/Arduino.mk:1600: recipe for target 'reset' failed
make[1]: *** [reset] Error 127
make[1]: Leaving directory '/sonah/src/sleepypi/run/script2'
/sonah/src/sleepypi/run/script2/Arduino.mk:1558: recipe for target 'upload' failed
make: *** [upload] Error 
```

I would love to get some insight in what i did wrong and how i can fix it.

sej7278, 2018-04-25:
The Bare Arduino Project example makefile is a bit too specific, you shouldn't have to manually define most of those values. Try this project's examples instead. the problem is that ard-reset-arduino can't be found (does the sleepypi even use that?) it should be in your $PATH

zagatta-sonah, 2018-04-26:
@sej7278 
Which example do you mean? I am not sure if the SleepyPi uses ard-reset-arduino, in fact I interpret the documentation that a mod of avrdude is used: [link1](https://github.com/SpellFoundry/avrdude-rpi) [link2](https://spellfoundry.com/sleepy-pi/setting-arduino-ide-raspbian/#Setting_up_the_Reset_DTR_pin)

@ladislas 
Do you suggest to use your Bare-Arduino-Project instead of the project I am currently using? I will try it out, thx!

ladislas, 2018-04-26:
`ARDMK_DIR         = $(PROJECT_DIR)` might be the issue.

zagatta-sonah, 2018-04-26:
@ladislas Thanks! I was not aware that the Makefile was from your project. I just used the Arduino-Makefile/examples/MakefileExample/Makefile-example.mk

I'll try to remove the `ARDMK_DIR ` and see what comes up.

Thanks!

PS: My use of the phrase "are you suggesting" might have sound angry in english (german speaker here), which was unintentional.

ncm, 2018-12-24:
I get the same error building Makeblocks example programs.  I tried setting ARDMK_DIR, and not, and in different directories. The only thing that actually worked was the patch from Pull Request 337.

Given a choice between "-j" working and being able to complete a build, I will take a complete build every time.  Parallel make is not so important for arduino builds anyway -- they are done so quickly without.