sudar/Arduino-Makefile #565: Compiling SAM projects on Windows fails with chip.h: No such file or directory

michaelrommel, 2018-06-05:
I tried to compile a sketch for a Arduino Due today on a Windows/cygwin installation and it fails with:
```
C:/Users/xxxxx/AppData/Local/Arduino15/packages/arduino/hardware/sam/1.6.11/cores/arduino/Arduino.h:42:18: fatal error: chip.h: No such file or directory
 #include <chip.h>
                  ^
compilation terminated.
```
The overall installation works compiling AVR based sketches!
The Makefile has the following contents:

```
ARDUINO_DIR = D:/Arduino
ARDMK_DIR = D:/xxxx/Software/Arduino-Makefile
MONITOR_PORT = com3
MONITOR_CMD = putty
MONITOR_PARAMS = 8,1,n,N
MONITOR_BAUDRATE = -b 57600

# Project details
ARCHITECTURE = sam
BOARD_TAG = arduino_due_x

# Define Arduino support package installation path where SAM device support has been installed
# # Linux
# #ARDUINO_PACKAGE_DIR := $(HOME)/.arduino15/packages
# # macOS
# #ARDUINO_PACKAGE_DIR := $(HOME)/Library/Arduino15/packages
# # Windows
ARDUINO_PACKAGE_DIR := C:/Users/$(USER)/AppData/Local/Arduino15/packages
USER_LIB_PATH := C:/Users/xxxx/AppData/Local/Arduino15/packages/arduino/hardware/sam/1.6.11/system/libsam
USER_INCLUDES := C:/Users/xxxx/AppData/Local/Arduino15/packages/arduino/hardware/sam/1.6.11/system/libsam

include D:/xxxx/Software/Arduino-Makefile/Sam.mk
```

I know that the file chip.h is located in the `.../hardware/sam/1.6.11/system/libsam` directory, but I have been unable to make the Makefile find this directory and compile successfully. It fails whether I have those experimental lines with USER_LIB_PATH or USER_INCLUDES in the Makefile or not.

The Arduino IDE is version 1.8.5 and the SAM installation is 1.6.11.

Can you provide any pointers, where I might look? Thanks a lot for your help, I have now exhausted all ideas...

michaelrommel, 2018-06-05:
I have now spent several hours to try to fix this. The error occurred also on a Mac. I think the whole Sam.mk is not prepared to deal with the current directory structure of the sam board variant.

I patched Sam.mk and Arduino.mk and was able to come to a linker stage, but still the linking fails with
```
UARTClass.cpp:(.text._ZN9UARTClass4initEmm+0x68): undefined reference to `SystemCoreClock'
collect2: error: ld returned 1 exit status
make: *** [build-arduino_due_x/MZeroBlink.elf] Error 1
```

The changes to Arduino.mk were:

```
diff -uNr /usr/local/opt/arduino-mk/Arduino.mk /Volumes/Data/Software/michael/arduino/Arduino-Makefile/Arduino.mk
--- /usr/local/opt/arduino-mk/Arduino.mk        2018-06-01 20:26:17.000000000 +0200
+++ /Volumes/Data/Software/michael/arduino/Arduino-Makefile/Arduino.mk  2018-06-05 23:20:41.000000000 +0200
@@ -846,6 +846,11 @@
             ALT_CORE_OBJ_FILES  = $(ALT_CORE_C_SRCS:.c=.c.o) $(ALT_CORE_CPP_SRCS:.cpp=.cpp.o) $(ALT_CORE_AS_SRCS:.S=.S.o)
             OTHER_OBJS   := $(patsubst $(ALTERNATE_CORE_PATH)/variants/$(VARIANT)/%,  \
                 $(OBJDIR)/core/%,$(ALT_CORE_OBJ_FILES))
+           # additional source files for sam
+            ALT_LIBSAM_OBJ_FILES  = $(ALT_LIBSAM_C_SRCS:.c=.c.o)
+            OTHER_OBJS   += $(patsubst $(ALTERNATE_CORE_PATH)/system/libsam/source/%,  \
+                $(OBJDIR)/core/%,$(ALT_LIBSAM_OBJ_FILES))
+            $(call show_config_variable,OTHER_OBJS,[DETECTED])
         endif

         CORE_OBJ_FILES  = $(CORE_C_SRCS:.c=.c.o) $(CORE_CPP_SRCS:.cpp=.cpp.o) $(CORE_AS_SRCS:.S=.S.o)
@@ -1009,6 +1014,7 @@

 ifdef ARDUINO_PLATFORM_LIB_PATH
     PLATFORM_INCLUDES     := $(foreach lib, $(PLATFORM_LIBS), $(call get_library_includes,$(lib)))
+    LIBSAM_INCLUDES       := -I$(wildcard $(ALTERNATE_CORE_PATH)/system/libsam/include)
     PLATFORM_LIB_CPP_SRCS := $(foreach lib, $(PLATFORM_LIBS), $(call get_library_files,$(lib),cpp))
     PLATFORM_LIB_C_SRCS   := $(foreach lib, $(PLATFORM_LIBS), $(call get_library_files,$(lib),c))
     PLATFORM_LIB_AS_SRCS  := $(foreach lib, $(PLATFORM_LIBS), $(call get_library_files,$(lib),S))
@@ -1049,7 +1055,7 @@
 # Using += instead of =, so that CPPFLAGS can be set per sketch level
 CPPFLAGS      += -$(MCU_FLAG_NAME)=$(MCU) -DF_CPU=$(F_CPU) -DARDUINO=$(ARDUINO_VERSION) $(ARDUINO_ARCH_FLAG) \
         -I$(ARDUINO_CORE_PATH) -I$(ARDUINO_VAR_PATH)/$(VARIANT) \
-        $(SYS_INCLUDES) $(PLATFORM_INCLUDES) $(USER_INCLUDES) -Wall -ffunction-sections \
+        $(SYS_INCLUDES) $(PLATFORM_INCLUDES) $(LIBSAM_INCLUDES) $(USER_INCLUDES) -Wall -ffunction-sections \
         -fdata-sections

 # PROG_TYPES_COMPAT is enabled by default for compatibility with the Arduino IDE.
@@ -1337,6 +1343,10 @@
        @$(MKDIR) $(dir $@)
        $(CC) -MMD -c $(CPPFLAGS) $(ASFLAGS) $< -o $@

+$(OBJDIR)/core/%.c.o: $(ALTERNATE_CORE_PATH)/system/libsam/source/%.c $(COMMON_DEPS) | $(OBJDIR)
+       @$(MKDIR) $(dir $@)
+       $(CC) -MMD -c $(CPPFLAGS) $(CFLAGS) $< -o $@
+
 # various object conversions
 $(OBJDIR)/%.bin: $(OBJDIR)/%.elf $(COMMON_DEPS)
        @$(MKDIR) $(dir $@)
```
and for Sam.mk
```
diff -uNr /usr/local/opt/arduino-mk/Sam.mk /Volumes/Data/Software/michael/arduino/Arduino-Makefile/Sam.mk
--- /usr/local/opt/arduino-mk/Sam.mk    2018-06-01 20:26:17.000000000 +0200
+++ /Volumes/Data/Software/michael/arduino/Arduino-Makefile/Sam.mk      2018-06-05 22:33:00.000000000 +0200
@@ -76,14 +76,16 @@
 endif

 ifndef CMSIS_DIR
-    CMSIS_DIR := $(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/tools/CMSIS/$(CMSIS_VER)/CMSIS
+    #CMSIS_DIR := $(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/tools/CMSIS/$(CMSIS_VER)/CMSIS
+    CMSIS_DIR := $(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/hardware/$(ARCHITECTURE)/$(CORE_VER)/system/CMSIS/CMSIS
     $(call show_config_variable,CMSIS_DIR,[AUTODETECTED],(from ARDUINO_PACKAGE_DIR))
 else
     $(call show_config_variable,CMSIS_DIR,[USER])
 endif

 ifndef CMSIS_ATMEL_DIR
-    CMSIS_ATMEL_DIR := $(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/tools/CMSIS-Atmel/$(CMSIS_ATMEL_VER)/CMSIS
+    #CMSIS_ATMEL_DIR := $(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/tools/CMSIS-Atmel/$(CMSIS_ATMEL_VER)/CMSIS
+    CMSIS_ATMEL_DIR := $(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/hardware/$(ARCHITECTURE)/$(CORE_VER)/system/CMSIS/
     $(call show_config_variable,CMSIS_ATMEL_DIR,[AUTODETECTED],(from ARDUINO_PACKAGE_DIR))
 else
     $(call show_config_variable,CMSIS_ATMEL_DIR,[USER])
@@ -112,6 +114,7 @@
 # add CMSIS includes
 CPPFLAGS += -I$(CMSIS_DIR)/Include/
 CPPFLAGS += -I$(CMSIS_ATMEL_DIR)/Device/ATMEL
+CPPFLAGS += -I$(ALTERNATE_CORE_PATH)/system/libsam
 # path for Cortex library
 LIB_PATH  =  $(CMSIS_DIR)/Lib/GCC
 BOOTLOADER_PARENT = $(ALTERNATE_CORE_PATH)/bootloaders
@@ -131,7 +134,8 @@
 # grab any sources in the variant core path (variant.cpp defines pin/port mapping on SAM devices)
 ALT_CORE_C_SRCS := $(wildcard $(ALTERNATE_CORE_PATH)/variants/$(VARIANT)/*.c)
 ALT_CORE_CPP_SRCS := $(wildcard $(ALTERNATE_CORE_PATH)/variants/$(VARIANT)/*.cpp)
-ALT_CORE_S_SRCS := $(wildcard $(ALTERNATE_CORE_PATH)/variants/$(VARIANT)/*.S)
+ALT_CORE_AS_SRCS := $(wildcard $(ALTERNATE_CORE_PATH)/variants/$(VARIANT)/*.S)
+ALT_LIBSAM_C_SRCS := $(wildcard $(ALTERNATE_CORE_PATH)/system/libsam/source/*.c)

 # Use arm-toolchain from Arduino install if exists and user has not defined global version
 ifndef ARM_TOOLS_DIR
```

If anyone figures out, why the linker fails, I would be grateful. For now, since I want to work on a project with AVR and SAM devices and work on macOS as well as Windows, it seems, that I have to use the official Arduino GUI to get the job done. Sad, but true... My time is running out, I cannot spend more evenings debugging this. :-(

tuna-f1sh, 2018-06-06:
You're right, some work is required for Arduino Due support. The SAM support was developed for the SAMD Arduino microprocessors (Zero, M0, Feather, Sparkfun SAMD) - this should probably be made more clear in the README.

I've been meaning to get around to looking at the Due build process and adding support but I don't have one of these boards so can't physically test it and have also not had a need. Anything you can contribute would be useful.

tuna-f1sh, 2018-06-06:
I had a bit of time so gave this and stab and got a compiling branch. The SAM core less organised than the SAMD core and more reliant on the CMSIS folder structure. The problem you faced is that unlike the SAMD core, which has a startup.c with system init functions in the core folder, the SAM core must pull it from the CMSIS chip specific path.

It's a bit rough and I might make some changes for readability before a pull request but can you test it with hardware? Here is my branch: https://github.com/tuna-f1sh/Arduino-Makefile/tree/due

michaelrommel, 2018-06-06:
@tuna-f1sh 

John, that is amazing. It compiled on my Mac on the first try. Uploading failed, but I figured out that the "reset" command needs a full path `/dev/cu.usbmodemFA2311` and the bossac programmer needs just `cu.usbmodemFA2311`. I think I can figure out, where to put that in the Makefile.

Then there were a few errors with basename, but that might have to do with the macOS dialect, I'll look into it too, but it doesn't affect the compile!

Great job - I had given up hope! If I can get it to working completely this evening, that's worth spend instead of working in the Java GUI...

I'll keep posting here!

michaelrommel, 2018-06-06:
It just needed a small change
```
BOSSA_OPTS += --port=$(subst /dev,,$(call get_monitor_port))
```

tuna-f1sh, 2018-06-06:
Great. I'll add that if using Due (or I wonder if that is a problem on other boards that use Bossac too?) and make some changes to the README so that it's less Cortex-M0 specific, then make a pull request.

michaelrommel, 2018-06-06:
So, now I found the last error with the grep command, this section needs to be changed:
```
ifndef CXX_NAME
    CXX_NAME := $(call PARSE_BOARD,$(BOARD_TAG),build.command.g\+\+)
    ifndef CXX_NAME
        CXX_NAME := arm-none-eabi-g++
    else
        $(call show_config_variable,CXX_NAME,[COMPUTED])
    endif
endif
```

The error lies in the unescaped `+` signs of the `g++` pattern we are searching for.

michaelrommel, 2018-06-06:
@tuna-f1sh maybe you can correct the errors before you make the PR? I am able to track down the errors, but - for instance with the empty / not found paths - I am not really good at fixing things... You could see that in my feeble attempts to try to patch the Makefile...

Good thing, though - I read a lot in the GNU make manual and learned already a lot...

michaelrommel, 2018-06-06:
Hi, I just tested some examples and the compiled files do not work on the Due :-( I will try to compare the binaries. For the same basic blink sketch with Serial output the file sizes differ significantly. The Arduino-Makefile version is around 13k and the Arduino GUI version around 24k.

tuna-f1sh, 2018-06-06:
Are you using the toolchain included with Arduino or system installed? I
have found that using the latest gcc compiler can cause a sketch not to
work with the arm core. First thing to check would be this then maybe
optimisation flags if the build is much smaller.

And glad you’re picking stuff up. I have done the same and why I’ve enjoyed
contributing to this project.

michaelrommel, 2018-06-06:
The included toolchain gets used. I compared now all the parameters in the linking stage and they are mostly the same. Biggest difference is the inclusion of one static library in to the linking process:

`/Volumes/Data/Users/rommel/Library/Arduino15/packages/arduino/hardware/sam/1.6.11/variants/arduino_due_x/libsam_sam3x8e_gcc_rel.a`

And like this makefile, the Arduino GUI also compiles the core itself, but the commands are not shown in the debug output of the GUI. They are something like:

```
"/var/folders/t8/qnzcz5xs7nb969js5spr0xlh0000gn/T/arduino_build_570923/core/variant.cpp.o"
"/var/folders/t8/qnzcz5xs7nb969js5spr0xlh0000gn/T/arduino_build_570923/../arduino_cache_792092/core/core_arduino_sam_arduino_due_x_dbg_00efb0933f3483933de521f60e844971.a"
```

I first thought the bossac command line flags would make a difference, since the GUI adds two flags: `-U false` and `-b` but that didn't work.

tuna-f1sh, 2018-06-07:
Try my latest commit. I've added the 'libsam_3x8e' lib and also corrected the ARM math lib for the Due Cortex-M3.

michaelrommel, 2018-06-07:
Hi John, yes, now it works! The file size is aroud 24k like with the GUI.

But I needed to add `-U false -b` to the bossac command line. Without it the LEDs would just stay on and the sketch wouldn't work. Now also the Serial line outputs characters correctly!

Well done, John!! Thanks a lot for your support!!

tuna-f1sh, 2018-06-07:
Glad to hear it. I've added those flags to the `bossac` command now. Thanks for your clear and concise support too.

sej7278, 2018-06-22:
can this be closed now #567 got merged?

tuna-f1sh, 2018-06-23:
I believe so.

michaelrommel, 2018-06-25:
Dear all,

I have switched today from tuna-f1sh's branch back to the official one here. The project I am working on compiles now flawlessly on the Mac and uploading works as well. 

Thank you all for this amazing support!

  Michael.

closed 2018-06-25 by michaelrommel