sudar/Arduino-Makefile #568: Makefile not working with Arduino-1.8.5

freiser77, 2018-06-27:
I try to make an OttoDIY project with the Makefile. As ARDUINO_LIBS I've set Servo.

When trying to build it I get the following error message:

```
/home/pi/arduino-mk/Arduino.mk:1001: *** The following libraries specified in ARDUINO_LIBS could not be found (searched USER_LIB_PATH, ARDUINO_LIB_PATH and ARDUINO_PLATFORM_LIB_PATH): Servo. Stop.
```

I've set the ARDUINO_DIR in .zshrc. The board used is a Arduino Nano w/ 328P.

What can I do to get it running?

sej7278, 2018-06-27:
it does work with 1.8.5 so might be worth changing your title to refer to servo/ottodiy.

you shouldn't have to specify ARDUINO_LIBS at all, show us your Makefile and ardunio-mk output.

this should be all you need:

```make
ARDUINO_DIR = /where/you/installed/arduino-1.8.5
BOARD_TAG = nano
BOARD_SUB = atmega328
include /home/pi/arduino-mk/Arduino.mk
```

freiser77, 2018-06-28:
I found it! There was no `set encoding=utf-8` in .vimrc. This led to some cumbersome chars in the Makefile. Now everything is working.

closed 2018-06-28 by freiser77