sudar/Arduino-Makefile #574: Raspberry Pi mods needed

pup-tv, 2018-07-26:
It took a few bits to get Arduino-Makefile working on a Raspberry Pi.

1.) Needed to update/download a later version of Arduino. Repository has something like version 1.0.5. Version 1.8.5 is current.

2.) mmcu decoding did not work with arduino-1.8.5 board.txt file.
   Set MCU=___  (atmega2560 for example) 

3.) Avr upload parameters:
  As is was sending: -c -b -P ...
  Need something like: -c avrisp2 -b 115200 -P ...

Thanks for supporting open source!!!

sej7278, 2018-07-27:
the 1.0.5 arduino works fine, as long as you use the boards.txt that matches e.g. https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile#L13

pup-tv, 2018-07-27:
Agreed...  if you can use Arduino 1.0.5. 

The code I was using used a lot of new language features  that were not supported in 1.0.5 throwing up crazy errors in the arduino library...

sej7278, 2018-07-28:
ah well to use 1.8.5 you need a BOARD_SUB not just a BOARD_TAG e.g. https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile#L18

the raspberry pi is just another debian box, nothing special is needed unless you're using gpio for uploads or something.

pup-tv, 2018-07-29:
sej, that works like a charm!  Thanks!

closed 2018-07-29 by pup-tv