sudar/Arduino-Makefile #575: Suggestion: rewrite example files to use $(ARDMK_DIR) for finding Arduino.mk

alxpettit, 2018-07-28:
Since it appears that the makefile completely breaks if you don't edit ARDMK_DIR, the examples might as well simply pull from that rather than having the examples hardcode the include path.

sej7278, 2018-07-29:
something's wrong with your setup then, as i've never set ARDMK_DIR, its calculated from the include line e.g. /usr/share/arduino/Arduino.mk

sej7278, 2018-07-29:
yes some examples use the linux+packaged paths /usr/share/arduino/Arduino.mk and some use mac+git (also used for travis-ci iirc) so there's not much consistency, although as they're examples of how to do it across platforms i guess that's not a bad way. you could add comments to each example with how to do it using:

linux+package (e.g. debian, fedora, arch)
mac+git
cygwin+whatever

obviously messing with ARDMK_DIR is going to give you a world of pain as you local Makefile won't be able to find Arduino.mk (hence its better to hardcode the include, cough, ahem, wink....)

sej7278, 2020-09-16:
fixed in #92 ce088e097de11e48e4ad43e0d7f436ae6717dbbf

closed 2020-09-16 by sej7278