sudar/Arduino-Makefile #576: /usr/etc/avrdude.conf is not the standard path on Arch Linux

alxpettit, 2018-07-28:
Makefile only looks under `/usr/etc/avrdude.conf`. On Arch Linux, it installs under `/etc/avrdude.conf` (which to me seems more in keeping with standard directory paths on modern Linux systems).

This does not affect functionality in the Arduino IDE, but it breaks functionality of Arduino-Makefile on Arch as of commit #b2d1ff23a2ff6d824aced575401e7b0deace672c.

sej7278, 2018-07-29:
this is because the arch arduino* packages aren't done correctly, sounds like they're setting AVR_TOOLS_DIR to /usr ?

AVR_TOOLS_DIR on debian for example is set to /usr/share/arduino/hardware/tools/avr/ and avrdude.conf is looked for in:

/usr/share/arduino/hardware/tools/avr/../
/usr/share/arduino/hardware/tools/avr/etc/

which then have a symlink inside: avrdude.conf -> /etc/avrdude.conf

so this matches the simply unzipped ide from arduino which has avrdude.conf in hardware/tools/avr/etc/avrdude.conf

you're welcome to make a PR that simply adds /etc/avrdude.conf to the searched locations, as you say that's where it will be on most distro's.

UltraBlackLinux, 2021-08-05:
I'd also like to have this fixed. Same problem currently