sudar/Arduino-Makefile #581: Sam.mk

thalesmaoa, 2018-09-19:
Compilers and plugin are bugged.

Just installed SAM board

```
./arduino --install-boards arduino:sam
```
Downloaded arduino-1.8.7

```
# Arduino Due uses SAM3X8E SAM chip
BOARD_TAG = arduino_due_x
ARCHITECTURE = sam
ARDUINO_DIR = /home/user/NAND/arduino-1.8.7
include ../Arduino-Makefile/Sam.mk
```
make complains about missing arm-none-eabi-gcc and arm-none-eabi-ar. I managed to workaround creating a symlink to arm-none-eabi-gcc-4.8.3 and arm-none-eabi-gcc-ar.

However, it is now complaining about missing `liblto_plugin.so`. However, it is insude:
```
arduino-1.8.7/hardware/tools/avr/libexec/gcc/avr/5.4.0/liblto_plugin.so
```

No clue how to point to there.

tuna-f1sh, 2018-09-19:
I haven't used the latest Arduino IDE with command line tools (which it looks like you are using here). I'm hoping they haven't changed the install location of support packages but they may have.

You need to define ARDUINO_PACKAGE_DIR ([as explained here](https://github.com/sudar/Arduino-Makefile#arduino-arm-boards) if the location is non-default (or you might have a .arduino15 folder in your home path already which it will [default to](https://github.com/sudar/Arduino-Makefile/blob/master/Sam.mk#L34)). 

Bascially, find where where the board support has been installed to (probably /home/user/NAND/arduino-1.8.7/packages) and define that as ARDUINO_PACKAGE_DIR. That `liblto_plugin.so` is an AVR shared object so is not correct.

I'll have a look at full compatibility with the Arduino CLI when I have time.

thalesmaoa, 2018-09-19:
I think ARDUINO_PACKAGE_DIR is right.

Please check the output:

```
$ make
-------------------------
Sam.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/thales/NAND/arduino-1.8.7 
- [AUTODETECTED]       ARDUINO_PACKAGE_DIR = /home/thales/.arduino15/packages (from DEFAULT)
- [AUTODETECTED]       CORE_VER = 1.6.11 (from ARDUINO_PACKAGE_DIR)
- [AUTODETECTED]       CMSIS_DIR = /system/CMSIS/CMSIS (from ARDUINO_PACKAGE_DIR)
- [AUTODETECTED]       CMSIS_ATMEL_DIR = /system/CMSIS (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           ARM_TOOLS_DIR = /home/thales/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/thales/.arduino15/packages/arduino/hardware/sam/1.6.11/libraries (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           UPLOAD_TOOL = bossac 
- [AUTODETECTED]       BOSSA = /home/thales/.arduino15/packages/arduino/tools/bossac/1.6.1-arduino/bossac (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           USB_PRODUCT = "Arduino Due" 
- [COMPUTED]           USB_MANUFACTURER = "Arduino LLC" 
- [COMPUTED]           USB_VID = 0x2341 
- [COMPUTED]           USB_PID = 0x003e 
-------------------------
Arduino.mk Configuration:
- [USER]               ARDMK_DIR = /home/thales/NAND/Arduino-Makefile 
- [AUTODETECTED]       ARDUINO_VERSION = 187 
- [USER]               ARCHITECTURE = sam 
- [USER]               ARDMK_VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK =  
- [BUNDLED]            AVR_TOOLS_DIR = /home/thales/NAND/arduino-1.8.7/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /home/thales/NAND/arduino-1.8.7/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_PLATFORM_LIB_PATH = /home/thales/.arduino15/packages/arduino/hardware/sam/1.6.11/libraries 
- [USER]               ALTERNATE_CORE_PATH = /home/thales/.arduino15/packages/arduino/hardware/sam/1.6.11 
- [COMPUTED]           ARDUINO_VAR_PATH = /home/thales/.arduino15/packages/arduino/hardware/sam/1.6.11/variants (from ALTERNATE_CORE_PATH)
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = arduino_due_x 
- [COMPUTED]           CORE = arduino (from build.core)
- [USER]               VARIANT = arduino_due_x 
- [COMPUTED]           OBJDIR = build-arduino_due_x (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = /home/thales/.arduino15/packages/arduino/hardware/sam/1.6.11/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [USER]               MCU_FLAG_NAME = mcpu 
- [USER]               CFLAGS_STD = -std=gnu99 
- [DEFAULT]            CXXFLAGS_STD =  
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyACM0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
- [USER]               BOOTLOADER_PARENT = /home/thales/.arduino15/packages/arduino/hardware/sam/1.6.11/bootloaders 
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 4.8.3 (arm-none-eabi-gcc)
-------------------------
mkdir -p build-arduino_due_x
/home/thales/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-ar rcs build-arduino_due_x/libcore.a  build-arduino_due_x/core/variant.cpp.o  build-arduino_due_x/core/tc.c.o  build-arduino_due_x/core/interrupt_sam_nvic.c.o  build-arduino_due_x/core/rtt.c.o  build-arduino_due_x/core/uotghs_device.c.o  build-arduino_due_x/core/uotghs.c.o  build-arduino_due_x/core/twi.c.o  build-arduino_due_x/core/adc.c.o  build-arduino_due_x/core/udphs.c.o  build-arduino_due_x/core/spi.c.o  build-arduino_due_x/core/efc.c.o  build-arduino_due_x/core/dacc.c.o  build-arduino_due_x/core/pio.c.o  build-arduino_due_x/core/pmc.c.o  build-arduino_due_x/core/emac.c.o  build-arduino_due_x/core/adc12_sam3u.c.o  build-arduino_due_x/core/uotghs_host.c.o  build-arduino_due_x/core/rstc.c.o  build-arduino_due_x/core/trng.c.o  build-arduino_due_x/core/can.c.o  build-arduino_due_x/core/pwmc.c.o  build-arduino_due_x/core/rtc.c.o  build-arduino_due_x/core/timetick.c.o  build-arduino_due_x/core/ssc.c.o  build-arduino_due_x/core/udp.c.o  build-arduino_due_x/core/usart.c.o  build-arduino_due_x/core/wdt.c.o  build-arduino_due_x/core/gpbr.c.o  build-arduino_due_x/core/system_sam3xa.c.o  build-arduino_due_x/core/wiring.c.o  build-arduino_due_x/core/WInterrupts.c.o  build-arduino_due_x/core/wiring_digital.c.o  build-arduino_due_x/core/itoa.c.o  build-arduino_due_x/core/hooks.c.o  build-arduino_due_x/core/syscalls_sam3.c.o  build-arduino_due_x/core/wiring_analog.c.o  build-arduino_due_x/core/wiring_shift.c.o  build-arduino_due_x/core/iar_calls_sam3.c.o  build-arduino_due_x/core/cortex_handlers.c.o  build-arduino_due_x/core/avr/dtostrf.c.o  build-arduino_due_x/core/IPAddress.cpp.o  build-arduino_due_x/core/wiring_pulse.cpp.o  build-arduino_due_x/core/new.cpp.o  build-arduino_due_x/core/WString.cpp.o  build-arduino_due_x/core/Stream.cpp.o  build-arduino_due_x/core/Print.cpp.o  build-arduino_due_x/core/watchdog.cpp.o  build-arduino_due_x/core/Reset.cpp.o  build-arduino_due_x/core/USARTClass.cpp.o  build-arduino_due_x/core/main.cpp.o  build-arduino_due_x/core/abi.cpp.o  build-arduino_due_x/core/RingBuffer.cpp.o  build-arduino_due_x/core/UARTClass.cpp.o  build-arduino_due_x/core/WMath.cpp.o  build-arduino_due_x/core/USB/USBCore.cpp.o  build-arduino_due_x/core/USB/PluggableUSB.cpp.o  build-arduino_due_x/core/USB/CDC.cpp.o  build-arduino_due_x/core/wiring_pulse_asm.S.o         
/home/thales/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-ar: Cannot find plugin 'liblto_plugin.so'
/home/thales/NAND/Arduino-Makefile/Arduino.mk:1562: recipe for target 'build-arduino_due_x/libcore.a' failed
make: *** [build-arduino_due_x/libcore.a] Error 1
```

thalesmaoa, 2018-09-20:
Ok. A bit more digging and found that all files are being compiled fine with the `ln` links, however the merge process is the problem. 

After many trials I decided to go through Microsoft solution. Delete everything and build a fresh install, and yeah... It works now!

Thanks

tuna-f1sh, 2018-09-20:
Glad you got it sorted, this is was I was going to suggest since it was finding the paths correctly and the `liblto_plugin.so` should have been part of the ARM toolchain installed with the package.

sej7278, 2018-09-20:
This worked fine for me, no errors at all:

```
./arduino --install-boards arduino:sam
```

Makefile:
```make
ARDUINO_DIR = $(HOME)/arduino-1.8.7
BOARD_TAG = arduino_due_x
ARCHITECTURE = sam

include $(HOME)/Arduino-Makefile/Sam.mk
```

thalesmaoa, 2018-09-27:
Just reading the issue again, just forgot to mention:

```
$ uname -a
Linux rasp 4.14.70+ #1144 Tue Sep 18 17:20:50 BST 2018 armv6l GNU/Linux
```

After I figure that was some installation problem, I encountered another problem:

```
ARD_RESET_OPTS += --zero
```
It seems that `ard-reset-arduino` for arm plataform doesn't like zero flag.
Besides, I also get an error when I try to upload. No device found, but monitor works fine.

I've changed the issue name.
Just using `x64` for solving this specific problem.

sej7278, 2018-09-28:
we probably do need to make a new release now we've merged the sam stuff

@maqifrnswa can we get a debian upload done perhaps?

tuna-f1sh, 2018-09-28:
It would be good I think. There a single line change I've identified using the GDB options that would be good to add before a new release. I'll make a pull request.

sej7278, 2018-10-22:
[here](https://www.the-jedi.co.uk/downloads/arduino/) are debian/fedora builds based on today's HEAD, there's no binaries so should work fine on rpi's (certainly avr used to, not tried sam)

tuna-f1sh, 2018-10-23:
Thanks. I’m currently away without a computer but I’ll check it out when
I’m back next week - I can check the sam stuff on RPi.

tuna-f1sh, 2018-11-22:
Finally got around to testing this on a RPi (3 B+) and the build works as expected :+1: