sudar/Arduino-Makefile #587: Arduino updated the nano bootloader and now I can't upload to my old nanos

TT-392, 2018-10-18:
Arduino updated the nano bootloader in the new update so I can now only update to my old arduino nano if I select processor:"ATmega328P (old bootloader)".
but as far as I know I there is no way to tell the makefile to use the old bootloader.

sej7278, 2018-10-19:
```
$ ARDUINO_QUIET=1 make show_boards|grep nano
nano                Arduino Nano

$ ARDUINO_QUIET=1 make show_submenu| grep nano
nano       atmega168       ATmega168
nano       atmega328       ATmega328P
nano       atmega328old    ATmega328P (Old Bootloader)
```

so, Makefile should look like:

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.7
BOARD_TAG = nano
#BOARD_SUB = atmega328
BOARD_SUB = atmega328old
include $(HOME)/Arduino-Makefile/Arduino.mk
```

Or even specify it manually by adding:
```make
BOOTLOADER_PARENT = $(HOME)/arduino-1.8.7/hardware/arduino/avr/bootloaders
BOOTLOADER_PATH   = atmega
BOOTLOADER_FILE   = ATmegaBOOT_168_atmega328.hex
```

sej7278, 2020-09-16:
closed as question answered

closed 2020-09-16 by sej7278