sudar/Arduino-Makefile #594: Different Serial Port description when using Arduino IDE (Arduino micro)

asmateus, 2018-12-18:
Hi, I have run into an issue that is not critical, but somehow bothersome. When compiling/uploading a `.ino` source file into my Arduino Micro using the official IDE, the serial port description for the device reads `Arduino Micro`. But if I use the makefile offered here I get a serial port description that reads `USB IO Board`. **Note:** there is no problem with the uploaded program.

Here is how I test serial port descriptions:

    python -c 'from serial.tools.list_ports import comports; print([p.description for p in comports()])'

Does this mean I am missing some configuration steps? or is it an known (non-priority) issue?
is it possible for me to modify this string (it would be awesome)?

Thank you.

asmateus, 2019-01-08:
I just found it. You can set a custom `usb product name` attaching the appropriate flag to `CPPFLAGS`:

    CPPFLAGS += -DUSB_PRODUCT='"My Name"'

Note the double pair of quotes, as `make` will strip the first ones.

sej7278, 2019-01-09:
yes its worth adding product to vid+pid for 32u4 devices at least

tuna-f1sh, 2019-01-10:
Cool. Can you confirm that my fork branch programs your Micro with the correct name please @asmateus ? There is no reason it should not as I've checked the comilation pulls the right value but worth not making any assumptions!

asmateus, 2019-01-10:
@tuna-f1sh I confirm product name is correct with the new patches. Tested it on a Micro. Feel free to close this issue when you see fit.

sej7278, 2019-01-10:
was just about to test on a sparkfun pro micro, but is this just a 1.5+ thing, as none of the 32u4 boards.txt i use have anything but vid/pid for usb....?

only the 1.8 core: https://github.com/sparkfun/Arduino_Boards/blob/master/sparkfun/avr/boards.txt#L97

tuna-f1sh, 2019-01-11:
Yes I've added the usb manufacturer in there too for which which use it: https://github.com/tuna-f1sh/Arduino-Makefile/commit/2700aeb1811f1e1934863933c477d75e20bd6d60 , otherwise it is "Undefined", which matches what will happen in the core for an undefined board: https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/USBCore.cpp#L46

The core will actually re-define the usb manufacturer a few lines down based on the VID, so I guess that is why it is not always in the boards.txt for the large contributors.

sej7278, 2019-01-11:
on ide 1.0.5 with an arduino leonardo, i get:

```
 -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_PRODUCT=' '-DUSB_MANUFACTURER="Unknown"'
```

and sparkfun pro micro:

```
-DUSB_VID=0x1B4F -DUSB_PID=0x9206 '-DUSB_PRODUCT=' '-DUSB_MANUFACTURER="Unknown"' 
```

note the quotes are wrong btw.

perhaps if we don't have a product or manufacturer, we shouldn't even add to the cppflags? or set product to "USB IO Board" as per the ide code you linked to.

tuna-f1sh, 2019-01-11:
Thanks for testing. I've added a defined check before appending the USB_PRODUCT and USB_MANUFACTURER and corrected quotes. https://github.com/tuna-f1sh/Arduino-Makefile/tree/master

sej7278, 2019-01-11:
ah yes, i now get just this on arduino leonardo and ide 1.0.5:

```
 -DUSB_VID=0x2341 -DUSB_PID=0x8036 
```

on ide 1.8.8 i get this using the sparkfun pro micro:

```
-DUSB_VID=0x1b4f -DUSB_PID=0x9206 
    -DUSB_PRODUCT='"SparkFun Pro Micro"' -DUSB_MANUFACTURER='"Unknown"' 
```

are those quotes still iffy?

tuna-f1sh, 2019-01-11:
Yes, it prevents the shell 'eating' the quotes. Without them, the define will be only words up to a space.

Apparently [its more portable](https://stackoverflow.com/questions/2410976/how-to-define-a-string-literal-in-gcc-command-line) to escape the quotes \"blah\", but that would require some more pre-processing of the boards.txt grep and seems unnecessary as `'""'` us supported by gcc.

sej7278, 2019-01-11:
ok fine, if @asmateus could possibly try it with his arduino micro, i'll merge it, works with my sf pro micro:

before:
```
[947301.687735] usb 1-4.4: New USB device found, idVendor=1b4f, idProduct=9205, bcdDevice= 0.01
[947301.687739] usb 1-4.4: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[947301.687742] usb 1-4.4: Product: Pro Micro 5V  
[947301.687744] usb 1-4.4: Manufacturer: SparkFun Electronics
```

after:
```
[947318.478464] usb 1-4.4: New USB device found, idVendor=1b4f, idProduct=9206, bcdDevice= 1.00
[947318.478469] usb 1-4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[947318.478471] usb 1-4.4: Product: SparkFun Pro Micro
[947318.478474] usb 1-4.4: Manufacturer: SparkFun
```

asmateus, 2019-01-11:
@sej7278 here is what I get:

before:

    [12038.366276] usb 2-7: New USB device found, idVendor=2341, idProduct=8037, bcdDevice= 1.00
    [12038.366283] usb 2-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [12038.366287] usb 2-7: Product: USB IO Board
    [12038.366292] usb 2-7: Manufacturer: Arduino LLC

after:

    [12255.412858] usb 2-7: New USB device found, idVendor=2341, idProduct=8037, bcdDevice= 1.00
    [12255.412867] usb 2-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [12255.412873] usb 2-7: Product: Arduino Micro
    [12255.412879] usb 2-7: Manufacturer: Arduino LLC

Seems OK to me.

sej7278, 2019-01-11:
ah ok - i see you had the default product before. thanks for testing @asmateus 

thanks for the PR @tuna-f1sh, merged

closed 2019-01-11 by sej7278