sudar/Arduino-Makefile #595: ardmk-init missing from brew installed package

alissa-huskey, 2018-12-21:
I installed Arduino-Makefile using MacOS homebrew. I went to find the ardmk-init script in $ARDMK_DIR/bin, but it's not there.

I went ahead and cloned the repo from git, so I've got the script now. But in case this was an oversight, I thought I'd let you know.

tuna-f1sh, 2018-12-21:
Yes, most the package installers (rightly) follow the releases. The last release was 1.5 years ago so it is quite out of date.

There was a push to get a release out and clearing out the outstanding PRs. Seems like a good time to issue a new version? @sudar @sej7278 ?

alissa-huskey, 2018-12-22:
That's what I thought too. But the installed version is 1.6.0. Isn't that the current version?

```
$ brew info arduino-mk
sudar/arduino-mk/arduino-mk: stable 1.6.0, HEAD
https://github.com/sudar/Arduino-Makefile
/usr/local/Cellar/arduino-mk/1.6.0 (62 files, 219KB) *
  Built from source on 2018-12-18 at 16:11:51
From: https://github.com/sudar/homebrew-arduino-mk/blob/master/arduino-mk.rb
==> Options
--HEAD
	Install HEAD version
==> Caveats
To use the Arduino-Makefile, please add the following into your per-project Makefile

    include /usr/local/opt/arduino-mk/Arduino.mk

Also make sure you have the necessary dependencies installed. Refer to the Requirements section of the Arduino Makefile Readme file at https://github.com/sudar/Arduino-Makefile#requirements to install the dependencies.
```

tuna-f1sh, 2018-12-23:
Correct, but a version is something that a maintainer chooses to release; it's not the HEAD of the repository. Have a look [here](https://github.com/sudar/Arduino-Makefile/releases) and you'll see 1.6.0 was issued in July 2017 - it's the master HEAD frozen at that time.

If you want to install the _development_ version with homebrew, you can use `brew install --HEAD arduino-mk`

sej7278, 2018-12-26:
we'll organise a release in the new year, still trying to find what happened to the debian packager @maqifrnswa you there?

The changes to the upstream debian packaging scripts that are required (as well as the source tarball and changelog) are:

```patch
diff --git a/debian/arduino-mk.manpages b/debian/arduino-mk.manpages
index bc96e23..1b53640 100644
--- a/debian/arduino-mk.manpages
+++ b/debian/arduino-mk.manpages
@@ -1 +1,3 @@
 ard-reset-arduino.1
+robotis-loader.1
+ardmk-init.1
diff --git a/debian/control b/debian/control
index 0fb7457..0643606 100644
--- a/debian/control
+++ b/debian/control
@@ -1,13 +1,13 @@
 Source: arduino-mk
 Section: electronics
-Priority: extra
+Priority: optional
 Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
 Uploaders: Scott Howard <showard@debian.org>, Simon John <git@the-jedi.co.uk>
 Build-Depends: debhelper (>= 9)
-Standards-Version: 3.9.8
+Standards-Version: 4.1.2
 Homepage: https://github.com/sudar/Arduino-Makefile
-Vcs-Git: https://alioth.debian.org/anonscm/git/debian-science/packages/arduino-mk.git
-Vcs-Browser: https://anonscm.debian.org/gitweb/?p=debian-science/packages/arduino-mk.git
+Vcs-Git: https://salsa.debian.org/science-team/arduino-mk.git
+Vcs-Browser: https://salsa.debian.org/science-team/arduino-mk
 
 Package: arduino-mk
 Architecture: all
```

Fedora 29 and Debian test builds are [here](https://www.the-jedi.co.uk/downloads/arduino/)