sudar/Arduino-Makefile #601: PARSE_BOARD chops off flags which contain '=' signs

beldenfox, 2019-03-20:


szeder, 2019-03-21:
I believe you are right and the parser should indeed include everything following the first '=' up to the end of the line.

This used to work for Teensy, but regressed in 2442daf (Moved the PARSE_BOARD macro from Arduino.mk to Common.mk., 2018-09-13) in PR #580.

(On a slightly related note: [`Sam.mk` still has its own `PARSE_BOARD` macro](https://github.com/sudar/Arduino-Makefile/blob/master/Sam.mk#L135))

@wingunder

tuna-f1sh, 2019-03-21:
Seems like it would be a good idea to fix this in Common.mk and remove the `IFNDEF PARSE_BOARD...` in Sam.mk since it is now in Common.mk. The `IFNDEF` should mean it is not overwritten in Sam.mk but removing it will make it cleaner/easier to maintain.

stv0g, 2019-05-03:
I created a PR for the proposed solution.

I also checked that `PARSE_BOARD` from `Sam.mk` and `Common.mk` are equivalent.

commit 6cf275f0 (2019-06-03): fix regression: PARSE_BOARD chops off flags which contain '=' signs (closes #601)
closed 2019-06-03 by sudar