sudar/Arduino-Makefile #602: Build return 0 flash size with ARM board(MKRzero)

yannS2016, 2019-03-22:
Goo day all!

I have done some work using the make file with AVR based board so far with great success. however, i have switched to ARM board( arduino mkrzero) for a staret. i have the tools and core correctly setup as i can build and upload fine om the arduino IDE.
This is the process i follow using the MzeroBlink example:

Note: build system: Linux Debian( rapsberry PI 2)
```
# Arduino M0 Pro Programming Port
BOARD_TAG = mkrzero
### Path to the Arduino-Makefile directory.
ARDMK_DIR         = ../..

CFLAGS_STD        = -std=gnu11
ARCHITECTURE = samd

# Define Arduino support package installation path where SAM device support has been installed
# Linux
 ARDUINO_PACKAGE_DIR := $(HOME)/.arduino15/packages
include ../../Sam.mk

```
this build as below:
```
Sam.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX
- [USER]               ARDUINO_DIR = /home/pi/iotivity-constrained/port/arduino/arduino-home
- [USER]               ARDUINO_PACKAGE_DIR = /home/pi/.arduino15/packages
- [AUTODETECTED]       CORE_VER = 1.6.20 (from ARDUINO_PACKAGE_DIR)
- [AUTODETECTED]       CMSIS_VER = 4.5.0 (from ARDUINO_PACKAGE_DIR)
- [AUTODETECTED]       CMSIS_ATMEL_VER = 1.1.0 (from ARDUINO_PACKAGE_DIR)
- [AUTODETECTED]       CMSIS_DIR = /home/pi/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS (from ARDUINO_PACKAGE_DIR)
- [AUTODETECTED]       CMSIS_ATMEL_DIR = /home/pi/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           ARM_TOOLS_DIR =  (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/pi/.arduino15/packages/arduino/hardware/samd/1.6.20/libraries (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           UPLOAD_TOOL = bossac
- [COMPUTED]           BOOTLOADER_UPLOAD_TOOL = openocd
- [AUTODETECTED]       OPENOCD = /home/pi/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static/bin/openocd -s /home/pi/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino6-static/share/openocd/scripts/ (from ARDUINO_PACKAGE_DIR)
- [AUTODETECTED]       BOSSA = /home/pi/.arduino15/packages/arduino/tools/bossac/1.7.0/bossac (from ARDUINO_PACKAGE_DIR)
- [COMPUTED]           USB_PRODUCT = "Arduino MKRZero"
- [COMPUTED]           USB_MANUFACTURER = "Arduino LLC"
- [COMPUTED]           USB_VID = 0x2341
- [COMPUTED]           USB_PID = 0x804f
-------------------------
Arduino.mk Configuration:
- [USER]               ARDMK_DIR = ../..
- [DEFAULT]            ARDUINO_VERSION = 100
- [USER]               ARCHITECTURE = samd
- [USER]               ARDMK_VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK =
- [USER]               ALTERNATE_CORE_PATH = /home/pi/.arduino15/packages/arduino/hardware/samd/1.6.20
- [COMPUTED]           ARDUINO_VAR_PATH = /home/pi/.arduino15/packages/arduino/hardware/samd/1.6.20/variants (from ALTERNATE_CORE_PATH)
- [USER]               AVR_TOOLS_DIR = /home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1
- [COMPUTED]           ARDUINO_LIB_PATH = /home/pi/iotivity-constrained/port/arduino/arduino-home/libraries (from ARDUINO_DIR)
- [USER]               ARDUINO_PLATFORM_LIB_PATH = /home/pi/.arduino15/packages/arduino/hardware/samd/1.6.20/libraries
- [DEFAULT]            USER_LIB_PATH = /libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = mkrzero
- [COMPUTED]           CORE = arduino (from build.core)
- [USER]               VARIANT = mkrzero
- [COMPUTED]           OBJDIR = build-mkrzero (from BOARD_TAG)
- [USER]               ARDUINO_CORE_PATH = /home/pi/.arduino15/packages/arduino/hardware/samd/1.6.20/cores/arduino
- [ASSUMED]            MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [USER]               MCU_FLAG_NAME = mcpu
- [USER]               CFLAGS_STD = -std=gnu11
- [DEFAULT]            CXXFLAGS_STD =
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyACM0
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: Basic (not AVR-aware)
- [USER]               BOOTLOADER_PARENT = /home/pi/.arduino15/packages/arduino/hardware/samd/1.6.20/bootloaders
- [COMPUTED]           ARDMK_VERSION = 1.6
- [COMPUTED]           CC_VERSION = 4.8.3 (arm-none-eabi-gcc)
```
the build completes without issues but the size command return something strange:
```
/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-size build-mkrzero/MZeroBlink.hex
   text    data     bss     dec     hex filename
      0   10328       0   10328    2858 build-mkrzero/MZeroBlink.hex
```
and obviously the code wont do a thing after upload.
Can anyone help in this regard?
Thanks

tuna-f1sh, 2019-03-25:
I believe the strangeness with the size return is actually due to it parsing the .hex, not the .elf - .elf includes all the symbols, sections, fuse, lock bit settings etc, whereas .hex only has the binary data. `size` has nothing more than this binary data so returns it all as flash '.data'.

If you do `/home/pi/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/binarm-none-eabi-size build-mkrzero/MZeroBlink.elf` you should get the full breakdown, eg:

```
text    data     bss     dec     hex filename
10104     256    1964   12324    3024 build-mzero_pro_bl_dbg/MZeroBlink.elf
```
Clearly it this behaviour should be changed and SAMD builds should pass the .elf to size rather than .hex.

As for why it not working after upload, I'm not sure about this. I'm not sure anyone has actually tried and tested a MKRZero, there could be some differences?

ladislas, 2019-03-25:
The `avr-size` provided by the Arduino app or if you compile binutils yourself has a patch that's a few years old to provide the size for avr boards.

The patch hasn't evolved in years and does not contain all the new boards/mcu's.

And alternative for avr is to use `avr-objdump` like this:

```
$ avr-objdump -Pmem-usage file.elf
```

[Explanation here](https://www.avrfreaks.net/comment/2446681#comment-2446681)

I can't test it for arm but if it works, I think it could be generalized in the Makefile to not be dependent on a `size` bin that's broken.

tuna-f1sh, 2019-03-25:
I'm confused, I think we're talking about different things. The `arm-none-eabi-size` tool works as the `avr-size` if you pass the .elf. The output in this issue and default for ARM boards is to pass the .hex, which `size` is unable to present a full breakdown because it's just a binary file with no sections.

I'm not sure why AVR builds pass the .elf to the `size` and ARM passes the .hex but should be a fairly simple fix.

```
# M Zero Blink
make size
/home/john/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-size build-mzero_pro_bl_dbg/MZeroBlink.hex
   text    data     bss     dec     hex filename
      0   10360       0   10360    2878 build-mzero_pro_bl_dbg/MZeroBlink.hex
```

```
# AVR Blink
make size
/usr/bin/avr-size --mcu=atmega328p -C --format=avr build-uno/Blink.elf
AVR Memory Usage
----------------
Device: atmega328p

Program:     920 bytes (2.8% Full)
(.text + .data + .bootloader)

Data:          9 bytes (0.4% Full)
(.data + .bss + .noinit)
```

tuna-f1sh, 2019-03-25:
Ah ok, I see sort of what you were getting at. ARM was falling back to 'Basic (not AVR-aware)' and so not getting the .elf file. I've made a [quick fix](https://github.com/tuna-f1sh/Arduino-Makefile/commit/9783e68bc874329d33122b0fbdf5f3ffe79383f6), which means arm-size is properly supported.

yannS2016, 2019-03-29:
Thanks Guy for your support!

Also @tuna-f1sh , the code actually compiles fined, my mistake with the small testing sample is that i assumed the MKR Zero board used LED 13 as the build in LAMP, which is not. after fixing that i got it to pass the simple test. will communicate more on this board as i go.

Tanks Again.

closed 2019-03-31 by yannS2016

sej7278, 2019-03-31:
@tuna-f1sh is your fix going to become a PR?