sudar/Arduino-Makefile #607: Arduino Nano Every Support

k-kleber, 2019-07-02:
Arduino just released its new Nano variant EVERY
Nano Every
It uses the ATMega4809 processor, but is not supported using arduino-mk yet.

If someone could add the compile files for it it would be amazing, as I could not figure out how to do it myself.

Thanks

sej7278, 2019-07-02:
this compiles for me if i download the boards manager package mentioned [here](https://www.arduino.cc/en/Guide/NANOEvery), not got a board to test uploading etc.

```make
ARDUINO_DIR = $(HOME)/arduino-1.8.9
BOARD_TAG = nona4809
ALTERNATE_CORE_PATH = $(HOME)/.arduino15/packages/arduino/hardware/megaavr/1.8.2
include $(HOME)/Arduino-Makefile/Arduino.mk
```

k-kleber, 2019-07-03:
Thanks fir the reply. this has already started me in the right direction.
My remaining problem is, however, that it does not find the HardwareSerial.h file, which is a dependencies of one of the libraries i am using. 

It is located in `~/arduino-1.8.9/hardware/arduino/avr/cores/arduino/HardwareSerial.h`

[arduino-make-out.txt](https://github.com/sudar/Arduino-Makefile/files/3354267/arduino-make-out.txt)

sej7278, 2019-07-03:
the api directory seems to be some new HAL feature in recent cores, also seems to have a weird boards.txt location https://github.com/arduino/ArduinoCore-megaAVR

i added this simple change to add the api subdirectory to the include path and it seems to compile HardwareSerial now:

```diff
diff --git a/Arduino.mk b/Arduino.mk
index 515f590..89683c7 100644
--- a/Arduino.mk
+++ b/Arduino.mk
@@ -1159,7 +1159,7 @@ endif
 
 # Using += instead of =, so that CPPFLAGS can be set per sketch level
 CPPFLAGS      += -$(MCU_FLAG_NAME)=$(MCU) -DF_CPU=$(F_CPU) -DARDUINO=$(ARDUINO_VERSION) $(ARDUINO_ARCH_FLAG) \
-        -I$(ARDUINO_CORE_PATH) -I$(ARDUINO_VAR_PATH)/$(VARIANT) \
+        -I$(ARDUINO_CORE_PATH) -I$(ARDUINO_CORE_PATH)/api -I$(ARDUINO_VAR_PATH)/$(VARIANT) \
         $(SYS_INCLUDES) $(PLATFORM_INCLUDES) $(USER_INCLUDES) -Wall -ffunction-sections \
         -fdata-sections
 ```

k-kleber, 2019-07-03:
Perfect!!! This solved it.
Thanks for the help :)

sej7278, 2019-07-03:
ok thanks for confirming, made PR #609

closed 2019-07-03 by sej7278

sej7278, 2019-07-03:
updated my pr to also compile c/cpp files in the api directory as per PR #599

eudoxos, 2020-05-20:
This might be useful for someone wresting with Nano Very via Makefiles. I had to customize the following (the first line is already mentioned):
```
BOARD_TAG=nona4809
ALTERNATE_CORE_PATH=$(HOME)/.arduino15/packages/arduino/hardware/megaavr/1.8.6
# needed for proper reset, otherwise avrdude gets stuck with
#    avrdude: jtagmkII_getsync(): sign-on command: status -1
ARD_RESET_OPTS=--caterina 
# for SPI and others
ARDUINO_PLATFORM_LIB_PATH=$(ALTERNATE_CORE_PATH)/libraries 
```
Arduino IDE is also setting extra flags after the sketch upload, but I am not sure how important these are (and I did not find a variable which lets me append those):
```
 -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m 
```

I tried adding those manually but still not the same result as with the IDE. The IDE uploads successfully and the board runs. Manual upload (with or without the fuses) uploads but then the board is stuck and does not even react to reset.

Anyone can shed light on that? Perhaps there is more of us with similar issue.

sej7278, 2020-05-20:
i'd be pretty surprised if the IDE is setting fuses when you do an upload, i'm also pretty surprised that its using the catalina reset method - that's usually leonardo models not nano (although this seems to be some odd mix of samd11 + emulated 328p)

eudoxos, 2020-05-21:
I don't have much of an explanation for the reset, though the default was consistently causing the upload error for me and `--caterina` did not; the IDE says before upload (if you can figure out which type of reset that implies):
```
Forcing reset using 1200bps open/close on port /dev/ttyACM1
```

Here is the working cut&paste avrdude line from the IDE:
```
/home/eudoxos/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/home/eudoxos/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -V -patmega4809 -cjtag2updi -P/dev/ttyACM1 -b115200 -e -D -Uflash:w:/tmp/arduino_build_138575/fotobox.ino.hex:i -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m {upload.extra_files}
```

and here is the failing command produced by the make:

```
/home/eudoxos/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -v -V -p atmega4809 -C /home/eudoxos/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -D -c jtag2updi -b 115200 -P /dev/ttyACM1 \
                -U flash:w:build-nona4809/fotobox.hex:i
```

sej7278, 2020-05-21:
looks like it is setting fuses, using some sort of jtag protocol to upload, that seems to be the only difference between the arduino-mk way and the ide

the 1200bps open/close serial is the caterina method yes.

himadrir, 2022-11-21:
Hi I am using ros serial with one of the arduino nano variant that uses the LGT8F328P chip instead of the 328p chip. I am facing connection issues. The rosserial uses the HardwareSerial.h that comes by default on arduino and I am not able to use the HardwareSerial.h provided for my nano. Can someone explain the steps to compile and solve this problem?