sudar/Arduino-Makefile #610: Windows/Cygwin error on make reset: no such file or directory

TaylorZowtuk, 2019-07-07:
Im new to this project and im having some trouble getting the blink example compiled and uploaded for a windows/cygwin setup. Here is the makefile im attempting to use:

```
# Arduino Make file. Refer to https://github.com/sudar/Arduino-Makefile

ARDUINO_DIR   = D:/Arduino
BOARD_TXT	= D:/Arduino/hardware/arduino/avr
BOARD_TAG    = uno
include ../../Arduino.mk

# --- leonardo (or pro micro w/leo bootloader)
#BOARD_TAG    = leonardo
#MONITOR_PORT = /dev/ttyACM0
#include /usr/share/arduino/Arduino.mk

# --- mega2560 ide 1.0
#BOARD_TAG    = mega2560
#ARDUINO_PORT = /dev/ttyACM0
#include /usr/share/arduino/Arduino.mk

# --- mega2560 ide 1.6
#BOARD_TAG    = mega
#BOARD_SUB    = atmega2560
#MONITOR_PORT = /dev/ttyACM0
#ARDUINO_DIR  = /where/you/installed/arduino-1.6.5
#include /usr/share/arduino/Arduino.mk

# --- nano ide 1.0
#BOARD_TAG    = nano328
#MONITOR_PORT = /dev/ttyUSB0
#include /usr/share/arduino/Arduino.mk

# --- nano ide 1.6
#BOARD_TAG   = nano
#BOARD_SUB   = atmega328
#ARDUINO_DIR = /where/you/installed/arduino-1.6.5
#include /usr/share/arduino/Arduino.mk

# --- pro mini
#BOARD_TAG    = pro5v328
#MONITOR_PORT = /dev/ttyUSB0
#include /usr/share/arduino/Arduino.mk

# --- sparkfun pro micro
#BOARD_TAG         = promicro16
#ALTERNATE_CORE    = promicro
#BOARDS_TXT        = $(HOME)/arduino/hardware/promicro/boards.txt
#BOOTLOADER_PARENT = $(HOME)/arduino/hardware/promicro/bootloaders
#BOOTLOADER_PATH   = caterina
#BOOTLOADER_FILE   = Caterina-promicro16.hex
#ISP_PROG          = usbasp
#AVRDUDE_OPTS      = -v
#include /usr/share/arduino/Arduino.mk

# --- chipkit
#BOARD_TAG = mega_pic32
#MPIDE_DIR = /where/you/installed/mpide-0023-linux64-20130817-test
#include /usr/share/arduino/chipKIT.mk

# --- pinoccio
#BOARD_TAG         = pinoccio256
#ALTERNATE_CORE    = pinoccio
#BOOTLOADER_PARENT = $(HOME)/arduino/hardware/pinoccio/bootloaders
#BOOTLOADER_PATH   = STK500RFR2/release_0.51
#BOOTLOADER_FILE   = boot_pinoccio.hex
#CFLAGS_STD        = -std=gnu99
#CXXFLAGS_STD      = -std=gnu++11
#include /usr/share/arduino/Arduino.mk

# --- fio
#BOARD_TAG = fio
#include /usr/share/arduino/Arduino.mk

# --- atmega-ng ide 1.6
#BOARD_TAG    = atmegang
#BOARD_SUB    = atmega168
#MONITOR_PORT = /dev/ttyACM0
#ARDUINO_DIR  = /where/you/installed/arduino-1.6.5
#include /usr/share/arduino/Arduino.mk

# --- teensy3
#BOARD_TAG   = teensy31
#ARDUINO_DIR = /where/you/installed/the/patched/teensy/arduino-1.0.6
#include /usr/share/arduino/Teensy.mk

# --- mighty 1284p
#BOARD_TAG         = mighty_opt
#BOARDS_TXT        = $(HOME)/arduino/hardware/mighty-1284p/boards.txt
#BOOTLOADER_PARENT = $(HOME)/arduino/hardware/mighty-1284p/bootloaders
#BOOTLOADER_PATH   = optiboot
#BOOTLOADER_FILE   = optiboot_atmega1284p.hex
#ISP_PROG          = usbasp
#AVRDUDE_OPTS      = -v
#include /usr/share/arduino/Arduino.mk

# --- mightycore
#ALTERNATE_CORE = MightyCore
#BOARD_TAG = 1284
#MCU = atmega1284p
#BOARD_TAG = 644
#MCU = atmega644p
#VARIANT = standard
#F_CPU = 16000000L
#include /usr/share/arduino/Arduino.mk

# ---  megacore
#BOARD_TAG      = 128
#ALTERNATE_CORE = megacore
#F_CPU          = 16000000L
#include /usr/share/arduino/Arduino.mk

# --- sanguino 644p
#BOARD_TAG      = atmega644p
#ALTERNATE_CORE = sanguino
#ISP_PROG       = usbasp
#AVRDUDE_OPTS   = -v
#include /usr/share/arduino/Arduino.mk

# --- atmega328p on breadboard
#BOARD_TAG    = atmega328bb
#ISP_PROG     = usbasp
#AVRDUDE_OPTS = -v
#BOARDS_TXT   = $(HOME)/arduino/hardware/breadboard/boards.txt
#include /usr/share/arduino/Arduino.mk
```

and here is the console output of a `make upload`:

```
D:\Documents\CompSci\Arduino\Arduino-Makefile\examples\Blink>make upload
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = WINDOWS
- [USER]               ARDUINO_DIR = D:/Arduino
- [COMPUTED]           ARDMK_DIR = /cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 189
- [DEFAULT]            ARCHITECTURE = avr
- [DEFAULT]            ARDMK_VENDOR = arduino
- [DEFAULT]            ARDUINO_SKETCHBOOK = C:\Users\Zee/Documents/Arduino
- [COMPUTED]           ARDUINO_VAR_PATH = D:/Arduino/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = D:/Arduino/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [BUNDLED]            AVR_TOOLS_DIR = D:/Arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = D:/Arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = D:/Arduino/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = C:\Users\Zee/Documents/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh
- [USER]               BOARD_TAG = uno
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = standard (from build.variant)
- [COMPUTED]           OBJDIR = build-uno (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = D:/Arduino/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600
- [DEFAULT]            OPTIMIZATION_LEVEL = s
- [DEFAULT]            MCU_FLAG_NAME = mmcu
- [DEFAULT]            CFLAGS_STD = -std=gnu11
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11
- [AUTODETECTED]       DEVICE_PATH =
- [DEFAULT]            FORCE_MONITOR_PORT =
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = D:/Arduino/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6
- [COMPUTED]           CC_VERSION = 5.4.0 (avr-gcc)
-------------------------
mkdir -p build-uno
make reset
make[1]: Entering directory '/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/examples/Blink'
/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/bin/ard-reset-arduino
D:\Python 3.6.4\python.exe: can't open file '/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/bin/ard-reset-arduino': [Errno 2] No such file or directory
make[1]: *** [../../Arduino.mk:1737: reset] Error 2
make[1]: Leaving directory '/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/examples/Blink'
make: *** [../../Arduino.mk:1694: upload] Error 2 
```

Im thinking the problem is related to improper installation of the dependancies? I got cygwin and the make and Perl packages. I have a windows installation of python (which im thinking may be the problem though i dont want to install a cygwin package of python because i worry about conflicts with my windows installation) and used `pip install pyserial`. Im confused by what is meant by "Alternatively, if you have setup Cygwin to use a Windows Python installation, simply install using pip:" in the readme. What is meant by setup? Did i miss a step where i should be telling cygwin about my windows installation of python?

tuna-f1sh, 2019-07-07:
Your thinking is correct. See https://stackoverflow.com/questions/15334201/correct-path-usage-in-cygwin-difference-between-python-c-somefile-py-pyt

Basically, the Windows Python install cannot resolve cydrive paths - that's why you want the Cygwin Python. You should be able to install the Cygwin Python and not have it effect your Windows install; it will just be used in the Cygwin shell. You will save yourself pain the future too, if you plan to use Python from Cygwin.

TaylorZowtuk, 2019-07-07:
Thanks for the confirmation. I installed the cygwin python3 binary but im still getting the same error. 

``` mkdir -p build-uno
make reset
make[1]: Entering directory '/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/examples/Blink'
/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/bin/ard-reset-arduino
D:\Python 3.6.4\python.exe: can't open file '/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/bin/ard-reset-arduino': [Errno 2] No such file or directory
make[1]: *** [../../Arduino.mk:1737: reset] Error 2
make[1]: Leaving directory '/cygdrive/d/Documents/CompSci/Arduino/Arduino-Makefile/examples/Blink'
make: *** [../../Arduino.mk:1694: upload] Error 2
```
It seems that its still trying to use my windows installation of python and not the cygwin one. Any thoughts on what i need to do to force the make/cygwin to use only the cygwin python installation?

TaylorZowtuk, 2019-07-07:
So digging into Arduino.mk it contains the following for reset code:
https://github.com/sudar/Arduino-Makefile/blob/0e462fcff173c0e6963f3b23ecb5fbb2e59737e7/Arduino.mk#L853-L878

Im very unfamiliar with makefile syntax so forgive me but it looks like line 864 is checking the output of the which python command. For me i get this when inside the cygwin terminal: 
```
$ which python
/cygdrive/d/Python 3.6.4/python

Zee@MSI ~
$ which python3
/usr/bin/python3
```
I assume im entering the else in line 866 because its not finding the cygwin path that it wants but im not sure what line 867 is doing then...

tuna-f1sh, 2019-07-08:
The reset script calls [/usr/bin/env python](https://github.com/sudar/Arduino-Makefile/blob/master/bin/ard-reset-arduino#L1), which means on your system, the Windows python `which python` is being invoked. 

>It seems like its saying that i need to make sure that D:\cygwin\bin, which is currently included in my windows path, is the first directory in the path list? Does that mean any time i need to work with the windows version i need to edit my windows path so that the cygwin directory isnt at the top of the path anymore and vice versa if i want to work with the cygwin installation?

Correct, welcome to the pain of Windows and Cygwin installations! I would suggest using Windows Subsystem Linux (WSL) instead - it's a much better solution and what I have moved to when I need to use a Windows system. The snippet you shared has no influence on the script path resolution, only the port, which is different even still with a Cygwin Python so is irrelevant.

There is a lot of confusion here a out of the scope really of an issue with this project. You should be able to get it sorted however, if you get `which python` to result in `/usr/bin/python`.

TaylorZowtuk, 2019-07-13:
The problem is fixed if you download the cygwin python2 package instead of python27 or some other variant...

TaylorZowtuk, 2019-07-13:
Running into a new error with pyserial. With the following makefile settings:
```

ARDUINO_DIR   = D:/Arduino
BOARD_TXT	= D:/Arduino/hardware/arduino/avr
BOARD_TAG    = uno
MONITOR_PORT = com4
ARCHITECTURE = avr
include ../../Arduino.mk
```
I get the following error:
```
make reset
make[1]: Entering directory '/cygdrive/d/Documents/CompSci/UABiomed/exoskeleton/setup/Arduino-Makefile/examples/Blink'
/cygdrive/d/Documents/CompSci/UABiomed/exoskeleton/setup/Arduino-Makefile/bin/ard-reset-arduino  /dev/ttyS3
Traceback (most recent call last):
  File "/cygdrive/d/Documents/CompSci/UABiomed/exoskeleton/setup/Arduino-Makefile/bin/ard-reset-arduino", line 133, in <module>
    ser = serial.Serial(args.port[0], 115200)
  File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (22, 'Invalid argument')
make[1]: *** [../../Arduino.mk:1737: reset] Error 1
make[1]: Leaving directory '/cygdrive/d/Documents/CompSci/UABiomed/exoskeleton/setup/Arduino-Makefile/examples/Blink'
make: *** [../../Arduino.mk:1694: upload] Error 2
```

Any idea what the issue is here?

TaylorZowtuk, 2019-07-14:
As an update: a partner found that this issue is resolved when you first upload a blank sketch through the arduino IDE. So after uploading a blank sketch through the IDE you can do make upload and it will properly upload the blink sketch...

Does anyone have any idea why this is a problem?