sudar/Arduino-Makefile #611: SAMD device does not build without ARM_TOOLS_DIR set

hwestbrook, 2019-07-13:
On my Mac installation of Arduino and Arduino-mk, I could not get make to build. I had the following error:
```
/Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -I/Users/{username}/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/{username}/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL -DMD -DUSBCON '-DUSB_PRODUCT="Arduino Zero"' '-DUSB_MANUFACTURER="Arduino LLC"' -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -mthumb -nostdlib --param max-inline-insns-single=500 -fno-exceptions -Wl,-Map=build-arduino_zero_native/amb.map -mcpu=cortex-m0plus -DF_CPU=48000000L -DARDUINO=189 -DARDUINO_ARCH_SAMD -I/Users/{username}/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/cores/arduino -I/Users/{username}/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/cores/arduino/api -I/Users/{username}/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/variants/arduino_zero    -Wall -ffunction-sections -fdata-sections -Os -fno-rtti -fno-threadsafe-statics -std=gnu++11 -fpermissive -fno-exceptions  amb.ino -o build-arduino_zero_native/amb.ino.o
make: /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1: Permission denied
make: *** [build-arduino_zero_native/amb.ino.o] Error 1
```

I found this was due to AVR_TOOLS_DIR being automatically set to a weird value (an array of two values?):
```
- [USER] AVR_TOOLS_DIR = /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4
```

If I set it, as below, the file will build OK.
```
# Arduino Zero Native Port (should work with Feather, MKR1000 etc.)
# BOOTLOADER: The bootloader on these devices loaded when reset is pressed twice
# or the port is opened/closed at 1200 BAUD. If there is no program on the device,
# you may have to manually enter bootloader by toggling reset twice.
# see http://www.avdweb.nl/arduino/samd21/virus.html
BOARD_TAG = arduino_zero_native

# Define alternative core path if using another board supplier
#ALTERNATE_CORE_PATH = $(HOME)/Arduino/hardware/sparkfun/samd

# Define monitor port and isp port (bootloader port).
# Will automatically detect if Linux/macOS but MUST be defined on Windows
#MONITOR_PORT = com40 # CDC serial
#ISP_PORT = com39 # bootloader

# Define ARM toolchain dir if not using Arduino supplied
ARM_TOOLS_DIR = /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1

# Define Arduino support package installation path where SAM device support has been installed
# Linux
#ARDUINO_PACKAGE_DIR := $(HOME)/.arduino15/packages
# macOS
ARDUINO_PACKAGE_DIR := $(HOME)/Library/Arduino15/packages
# Windows
#ARDUINO_PACKAGE_DIR := "C:/Users/$(USER)/AppData/Local/Arduino15/packages"

# include ../../Sam.mk

include /usr/local/opt/arduino-mk/Sam.mk
```

sej7278, 2019-07-13:
it could be as its not in your $PATH or within your arduino distribution and Arduino.mk has got to line 577 without finding arm-none-eabi-gcc

more likely its because Sam.mk line 165 is looking for ```TOOL_PREFIX``` which isn't set until line 180 - could you try moving line 180 to around 162 and then not setting ```ARM_TOOLS_DIR```?

```diff
diff --git a/Sam.mk b/Sam.mk
index a014e63..9187ec8 100644
--- a/Sam.mk
+++ b/Sam.mk
@@ -161,6 +161,8 @@ ifeq ($(findstring arduino_due, $(strip $(VARIANT))), arduino_due)
 endif
 
 # Use arm-toolchain from Arduino install if exists and user has not defined global version
+TOOL_PREFIX = arm-none-eabi
+
 ifndef ARM_TOOLS_DIR
     ARM_TOOLS_DIR = $(call dir_if_exists,$(wildcard $(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/tools/$(TOOL_PREFIX)-gcc/*))
     $(call show_config_variable,ARM_TOOLS_DIR,[COMPUTED],(from ARDUINO_PACKAGE_DIR))
@@ -176,9 +178,6 @@ endif
 
 ########################################################################
 # command names
-
-TOOL_PREFIX = arm-none-eabi
-
 ifndef GDB_NAME
     GDB_NAME := $(call PARSE_BOARD,$(BOARD_TAG),build.command.gdb)
     ifndef GDB_NAME
```

hwestbrook, 2019-07-13:
That didn't seem to affect the problem. Same error:
```
/Users/hwestbrook/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 /Users/hwestbrook/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -I/Users/hwestbrook/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/hwestbrook/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL -DMD -DUSBCON '-DUSB_PRODUCT="Arduino Zero"' '-DUSB_MANUFACTURER="Arduino LLC"' -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -mthumb -nostdlib --param max-inline-insns-single=500 -fno-exceptions -Wl,-Map=build-arduino_zero_native/amb.map -mcpu=cortex-m0plus -DF_CPU=48000000L -DARDUINO=189 -DARDUINO_ARCH_SAMD -I/Users/hwestbrook/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/cores/arduino -I/Users/hwestbrook/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/cores/arduino/api -I/Users/hwestbrook/Library/Arduino15/packages/arduino/hardware/samd/1.8.3/variants/arduino_zero    -Wall -ffunction-sections -fdata-sections -Os -fno-rtti -fno-threadsafe-statics -std=gnu++11 -fpermissive -fno-exceptions  amb.ino -o build-arduino_zero_native/amb.ino.o
make: /Users/hwestbrook/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1: Permission denied
make: *** [build-arduino_zero_native/amb.ino.o] Error 1
```

The part I don't understand is this section:
```
/Users/hwestbrook/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 /Users/hwestbrook/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ 
```

If I remove the first part of that command (which is a directory), the command does not error.

One other item to note is I have to do the same thing with the BOSSA uploader, ala:
```
# Define ARM toolchain dir if not using Arduino supplied
ARM_TOOLS_DIR = /Users/hwestbrook/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1
BOSSA = /Users/hwestbrook/Library/Arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac
```

tuna-f1sh, 2019-07-15:
Are you sure you don't have another script defining `ARM_TOOLS_DIR` somewhere? 

The quote in your first post is actually suggesting Sam.mk detected it as defined by USER so nothing automatic is going on:

>- [USER] AVR_TOOLS_DIR = /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4

In fact, I'm confused about the Makefile you say works; {username} is not standard `make` syntax`?

># Define ARM toolchain dir if not using Arduino supplied
ARM_TOOLS_DIR = /Users/{username}/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1

hwestbrook, 2019-07-16:
I do not have `ARM_TOOLS_DIR` set as an env variable on my machine.

I removed my username from the first set of logs so it was generic. That's not the issue.

sej7278, 2019-07-16:
its probably because you have two versions of arm-none-eabi-gcc in your arduino package directory that are matching the wildcard, in which case you'd have to set it manually to select the one the makefile should use.

although that still doesn't explain why its display as user-set - or how it works at all when ```$(TOOL_PREFIX)``` isn't even defined until 15 lines later!

hwestbrook, 2019-07-16:
Feel free to close this issue if you want. Just wanted to log it in case someone else has something similar. My makefile is working fine with the variables specified explicitly. 

Thanks for making / supporting this tool!

tuna-f1sh, 2019-07-17:
I found this is an issue. #580 created the `TOOL_PREFIX`, added to the `ARM_TOOLS_DIR` auto resolve but it is defined after it! `ARM_TOOLS_DIR` was actually undefined in Sam.mk and Arduino.mk was auto-detecting `AVR_TOOLS_DIR`. The builds were still working because Arduino appears to install with arm-gcc in the AVR_TOOLS_DIR now. 

I've created a [fix in my fork](https://github.com/tuna-f1sh/Arduino-Makefile/commit/bdeaf1576be76beae3f223816692fd0dd6626f09), which also resolves the problem of multiple tools in the toolchain path; it will use the latest it multiple are found. If you can test it I'll create a pull request.

sej7278, 2019-07-17:
so your fork @tuna-f1sh will auto-select the highest version of gcc (lastword) if more than one is found and ```ARM_TOOLS_VER``` is not defined manually - and the same for bossac and openocd?

and you've moved ```TOOLS_PREFIX``` so its defined before use.

if you do a PR an update to HISTORY.md and [arduino-mk-vars.md](https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md#arm-variables) would be nice too please? ;-)

@hwestbrook could you test the fork please, I don't have multiple versions of the arm tools installed.