sudar/Arduino-Makefile #616: Fedora 30: Gives ambiguous python shebang when building package

GreyGnome, 2019-09-20:
When I am trying to build a package for Fedora 30, I get the following errors:

`*** ERROR: ambiguous python shebang in /usr/bin/ardmk-init: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/bin/robotis-loader: #!/usr/bin/python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/bin/ard-reset-arduino: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
`

To overcome this, I had to:

- tar xfz arduino-mk-1.6.0.tar.gz in the SOURCES directory

- cd arduino-mk-1.6.0/bin

- vim *

- replace all /usr/bin/env python with /usr/bin/env python2

- cd ../..

- tar cfz arduino-mk-1.6.0.tar.gz arduino-mk-1.6.0

- cd ../SPECS

Then I could do the rpmbuild -ba arduino-mk.spec

sej7278, 2019-09-20:
Have a look at my python3 PR

sej7278, 2020-09-16:
fixed in #e6881e2a43f02e0abf22bf28abc46b3c2e406d84

closed 2020-09-16 by sej7278