sudar/Arduino-Makefile #621: improve setup documentation to make Arduino-Makefile work on ubuntu/linux

LuisBL, 2019-11-25:
it took me some time to make this work on a recent ubuntu

LuisBL, 2019-11-25:
- install dependencies::

      $ sudo apt-get install python3-serial python-serial screen

- download
  - https://www.arduino.cc/download_handler.php?f=/arduino-1.8.10-linux64.tar.xz from https://www.arduino.cc/en/Main/Software
- Install arduino IDE::

      $ cd
      $ tar -xf arduino-1.8.10-linux64.tar.xz
      $ cd arduino-1.8.10; sudo sh install.sh

- Get Arduino-Makefile from github::

      $ git clone https://github.com/sudar/Arduino-Makefile ~/.arduino_mk

- Create a projet::

      $ mkdir ardui42; cd ardui42
      $ echo "ARDMK_DIR = \$(HOME)/.arduino_mk" > Makefile;  \
        echo "include \$(ARDMK_DIR)/Arduino.mk" >> Makefile; \
        echo "BOARD_TAG = uno" >> Makefile;                  \
        cp ~/.arduino_mk/examples/Blink/Blink.ino .

- Inform shell about ARDUINO_DIR::

      $ echo "export ARDUINO_DIR=/home/luis/arduino-1.8.10" >> ~/.bashrc
      $ source ~/.bashrc

- /dev/ttyACM0 should be read/write::

      $ sudo chmod 666 /dev/ttyACM0

- compile::

      $ make

- upload to your Arduino Uno (Plugged on USB port)::

      $ make upload

- On other terminale monitor serial communication (exit with Ctrl-a k [yes])::

      $ make monitor

  - on the terminal (a `screen` instance) we will see what we send to serial
    with `Serial.println("Hello foxes\r\n");`

LuisBL, 2019-11-25:
BTW thank you for this project, very useful.
If a you think that this could be useful I can make a patch and merge request to improve this.

sej7278, 2019-11-25:
you could have got most of that from [README.md](https://github.com/sudar/Arduino-Makefile/blob/master/README.md), unless you want bleeding edge you can simply do:

```sudo apt install arduino-mk python-serial arduino-core screen```

for the serial permissions, arduino used to have the instructions on their website, or even in the installer as i recall, you basically add your user to the dialout group, no need for chmod:

```sudo usermod -aG dialout yourusername```

logout and log back in again.

i use this in /etc/udev/rules.d/74-usbasp.rules for my ebay USBASP (all on one line) but i don't think you need to do this or chmod for arduino serial devices:

```SUBSYSTEMS=="usb", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="05dc", GROUP="dialout", MODE="0666"```

we do need to wake up the debian maintainer though as their package is pretty old, or figure out how to do a NMU.