sudar/Arduino-Makefile #622: Boards.txt has moved - how to resolve?

Tahutipai, 2019-12-02:
Running lastest version of Arduino at time of writing (1.8.10)

Make fails with `avr-g++: error: missing argument to ‘-mmcu=’`.
Reason: because it can't find `boards.txt`

It is looking for: `~/arduino/hardware/arduino/boards.txt`
But the new location is: `~/arduino/hardware/arduino/avr/boards.txt`

Is there an easy way around this?  Can we just point Arduino-Makefile at the new location?  Or is the only solution to install an older copy of Arduino (e.g 1.0.6) in /usr/share/arduino?

sej7278, 2019-12-02:
need more info - what board, what core, your Makefile etc.

its worth looking at the examples which have 1.0 and 1.6+ examples - you probably need to set a BOARD_SUB https://github.com/sudar/Arduino-Makefile/blob/master/examples/Blink/Makefile

sej7278, 2020-09-16:
closed as need more info

closed 2020-09-16 by sej7278