sudar/Arduino-Makefile #624: Computing of ARDUINO_VERSION leads to incompatibility with ATTinyCore

TLINDEN, 2020-02-18:
Hello,

the current master of Arduino-Makefile computes the variable `ARDUINO_VERSION` in a way which makes it hard to use it together with ATTinyCore (see: https://github.com/SpenceKonde/ATTinyCore/issues/389).

Here's the culprit:
```
Arduino.mk:294:    AUTO_ARDUINO_VERSION := $(shell [ -e $(VERSION_FILE) ] && cat $(VERSION_FILE) | sed -e 's/^[0-9]://g' -e 's/[.]//g' -e 's/\+.*//g' | head -c5)
```
With the latest Arduino verision this results in the number `1810`, which then gets compared to `10810`, which then fails.

I am not sure if this needs to be addressed here, or for what kind of things this variable is being used. If you want to change it, here's one way to do it:
```
echo 1.8.10 | awk -F\. '{ printf "%02d%02d%02d", $1, $2, $3 }'
010810
```

I myself fixed this by just setting the variable `ARDUINO_VERSION` in my Makefile to `010810`, see https://github.com/TLINDEN/attinycore-makefile-tests/commit/5e2e3603e96c27515ce4206c4051cd974f08c776.

And many thanks for the framework, I love it, since it allows me to use my beloved emacs!

ermyril, 2020-04-11:
+1, had the same issue when some library (midiusb) compared it with 10606 for arduino v1.6.6.

eharvester, 2020-05-19:
I have the same problem with other libraries. Should be 5-digits, for Reference pls. check:
https://github.com/arduino/arduino-cli/blob/master/docs/platform-specification.md#global-predefined-properties

As a workaround I set 'ARDUINO_VERSION = ddddd' (five digits) in my Makefile manually.

farblos, 2022-05-22:
Another symptom of this bug:

```
/home/farblos/work/arduino/libraries/NewliquidCrystal/I2CIO.cpp:34:29: fatal error: ../Wire/Wire.h: No such file or directory
compilation terminated.
```

`I2CIO.cpp` of my NewliquidCrystal distribution contains:

```
#if (ARDUINO < 10000)
   #include <../Wire/Wire.h>
#else
   #include <Wire.h>
#endif
```

which selects the wrong branch of the if statement since the Arduino makefile sed expression converts `1.8.13+dfsg1-2` from `version.txt` to four digit `1813`.

My sed expression to fix that first fills in the missing zeroes in one-digit version components (`s/\.\([0-9]\)\>/.0\1/g`), then strips the periods:

```
ARDUINO_VERSION :=	$(shell cat $(ARDUINO_DIR)/lib/version.txt | \
			        sed 's/^[0-9]://;s/\.\([0-9]\)\>/.0\1/g;s/\.//g;s/\+.*$$//')
```