sudar/Arduino-Makefile #625: Multiple toolchains cause makefile failure

dyntaos, 2020-02-23:
It appears that if more than one directories exist in the below directory, it causes ARM_TOOLS_DIR to contain multiple space separated paths and the makefile to fail

Sam.mk:165 `$(ARDUINO_PACKAGE_DIR)/$(ARDMK_VENDOR)/tools/$(TOOL_PREFIX)-gcc`
https://github.com/sudar/Arduino-Makefile/blob/e870443f4824cbbcd560f500c6072012dd668f62/Sam.mk#L165

I discovered this as my directory reference contained: "4.8.3-2014q1" and "7-2017q4"

dyntaos, 2020-02-24:
The same type issue applies to locating the bossac binary.

https://github.com/sudar/Arduino-Makefile/blob/e870443f4824cbbcd560f500c6072012dd668f62/Sam.mk#L274

More than one directory in `$(BUNDLED_BOSSA_DIR)` results in an invalid path in `$(BOSSA)`

My `$(BUNDLED_BOSSA_DIR)` contains: "1.7.0",  "1.7.0-arduino3"  and  "1.8.0-48-gb176eee"

tuna-f1sh, 2020-02-24:
I fixed this on my fork a while ago, I thought I'd made a pull request for it but must have forgotten: https://github.com/tuna-f1sh/Arduino-Makefile/commit/bdeaf1576be76beae3f223816692fd0dd6626f09

I'll make one now.

PR #640 (2020-09-03): Fix Travis CI, add SAMD test support, support GNU grep on macOS
Content as subject, includes fix #625. Includes @sej7278's Python 3 branch. BSD `grep` was creating some warnings parsing boards.txt that were annoying!
closed 2020-09-03 by sudar