sudar/Arduino-Makefile #629: avrdude port differs from $MONITOR_PORT

ermyril, 2020-04-11:
Hello!
When I'm trying to upload sketch using makefile, avrdude couldn't connect to the arduino, because I assume that it uses $MONITOR_PORT as a port in avrdude

make generates this command:
``` avrdude ...  -P /dev/cu.usbmodemMIDI1 ```
and then avr couldn't connect to the programmer

however, arduinoIDE passes that parameter like that
``` avrdude ...  -P /dev/cu.usbmodem1401 ``` 
and uploads successfully

```
ls /dev | grep cu.usbmodem
```
gives me only cu.usbmodemMIDI1

I tried to use $ARDUINO_PORT, setting $ISP_PORT and trying make ispload but it's not helping.

tuna-f1sh, 2020-04-14:
I don't understand the problem with Arduino Makefile here, the Arduino IDE is uploading successfully to a port that does not exist?! 

> ls /dev | grep cu.usbmodem
> gives me only cu.usbmodemMIDI1

If the correct port is `/dev/cu.usbmodem1401` you should be able to set it using `MONITOR_PORT = /dev/cu.usbmodem1401` but it isn't going to work if the port is not present!

ermyril, 2020-04-16:
Exactly, arduino IDE uses ```cu.usbmodem1401``` but only when it calls avrdude, in other places it uses ```usbmodemMIDI1```, and I believe that it IS the same port basically, because usbmodem1401 is not present on my system at all (hence makefile will fail immediately if I'll set ```MONITOR_PORT = /dev/cu.usbmodem1401```). I think that it's some strange encoding magic or so. It maybe some OS specific thing (I've got that error on macos mojave). 

Surely it's a very strange behaviour, but arduino IDE somehow aware of it. 

I'll post logs from makefile and from IDE when I'll get home. Also I'll try to test it on linux.

ladislas, 2020-04-16:
On macOS I use `MONITOR_PORT = /dev/cu.usbmodem*` or `MONITOR_PORT = /dev/tty.usbmodem*` and it works fine with avrdude.

tuna-f1sh, 2020-04-17:
What device are you using? It sounds like the bootloader is Catarina or similar whereby it enters a bootloader by toggling the baud rate. The port may enumerate as a different path once in the bootloader:

* `/dev/cu.usbmodem1401` is the bootloader port.
* `/dev/cu.usbmodemMIDI1` is the application port. 

Try using [ard-reset-arduino](https://github.com/sudar/Arduino-Makefile/blob/master/bin/ard-reset-arduino) with the `--catalina` flag and see whether usbmodem1401 appears.

If it does, add `ARD_RESET_OPTS += --caterina` to your Makefile.

ermyril, 2020-04-17:
@tuna-f1sh The device is a clone of an arduino pro micro.

Yup, caterina appears to be the case, when I've run ard-reset-arduino, ```usbmodemMIDI1 ``` dissapeared and```usbmodem1401``` showed in /dev/
added ```ARD_RESET_OPTS += --caterina```, but it generates ```/usr/local/bin/ard-reset-arduino --caterina --caterina /dev/cu.usbmodemMIDI1``` (so it was detected automatically) and still could not connect to the device.

@ladislas, thanks! I tried it but still with no luck :( 

Here is complete logs from running ```make upload``` - https://pastebin.com/u64hF5tF
Here is a log from arduino IDE - https://pastebin.com/7z8D99J0

And my makefile looks like this

```
ARDUINO_VERSION = 10801
ARDUINO_DIR   = /Applications/Arduino.app/Contents/Java
BOARD_TAG    = micro
MONITOR_PORT = /dev/tty.usbmodem*
ARD_RESET_OPTS += --caterina

include /usr/local/opt/arduino-mk/Arduino.mk
```

Alexlill, 2020-08-28:
I was torn between making a new issue for my case or necro-ing this thread, but I would have ended up with the exact same title and felt like it would have been spam. So, here I am:

Similarly to the OP, I'm ending up with the wrong ports relative to my MONITOR_PORT, my issue being that I'd like to make and upload the same .ino target for multiple different Arduinos.

Ex: 
I have verified that `ttyACM0` and `ttyACM1` are both present and function as intended.
My makefile: `MONITOR_PORT = /dev/ttyACM0` works as intended and uploads to the Arduino on port `ttyACM0`.
My makefile: `MONITOR_PORT = /dev/ttyACM1` eventually has the make output show `ard-reset-arduino  /dev/ttyACM0` and `avrdude ..... -P /dev/ttyACM0`, and thus fails to upload to the Arduino on `ttyACM1`.

I don't know if this is a common use case, but I thought I'd post anyways in case I was messing something up along the way.