sudar/Arduino-Makefile #635: Ambiguous python shebang, Fedora 31

Pertti2012, 2020-05-27:
Got following error when creating rpm for Fedora 31:

....
 + /usr/lib/rpm/brp-python-hardlink
+ /usr/lib/rpm/redhat/brp-mangle-shebangs
*** ERROR: ambiguous python shebang in /usr/bin/robotis-loader: #!/usr/bin/python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/bin/ard-reset-arduino: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
*** ERROR: ambiguous python shebang in /usr/bin/ardmk-init: #!/usr/bin/env python. Change it to python3 (or python2) explicitly.
error: Bad exit status from /var/tmp/rpm-tmp.LxKq4S (%install)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.LxKq4S (%install)
[ph@P2012 SPECS]$ cd ..

This is for your info, I fixed the problem in my env.
Br. Pertti Hyppönen

tuna-f1sh, 2020-05-27:
https://github.com/sudar/Arduino-Makefile/pull/620

sej7278, 2020-08-05:
Duplicate of #616

sej7278, 2020-09-16:
fixed in #e6881e2a43f02e0abf22bf28abc46b3c2e406d84

closed 2020-09-16 by sej7278