sudar/Arduino-Makefile #638: Another "avr-g++: error: missing device or architecture after '-mmcu='"

kzembower, 2020-06-04:
This error is very puzzling to me, as it was working perfectly for two days, then stopped. It's possible that I made a change in the Makefile inadvertently, but if so, I can't find it.

My Makefile is:
```
kevin@kevin-asrock:~/Arduino/Blink$ cat Makefile 
# Makefile for Arduino KitchenTimer project
# Author: Kevin Zembower
# Created: 2 June 2020

BOARD_TAG = pro
BOARD_SUB = 16MHzatmega328
ARDUINO_PORT = /dev/ttyUSB1
ARDUINO_LIBS = 
ARDUINO_DIR = /home/kevin/Arduino/arduino-1.8.12
AVR_TOOLS_DIR = /usr
AVRDUDE_CONF = /etc/avrdude.conf
include /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk
kevin@kevin-asrock:~/Arduino/Blink$
```

The BOARD and SUB choices are there:
```
kevin@kevin-asrock:~/Arduino/Blink$ ARDUINO_QUIET=1 make show_boards | grep Pro
pro                 Arduino Pro or Pro Mini
kevin@kevin-asrock:~/Arduino/Blink$ ARDUINO_QUIET=1 make show_submenu | grep pro
pro        16MHzatmega168  ATmega168 (5V, 16 MHz)
pro        16MHzatmega328  ATmega328P (5V, 16 MHz)
pro        8MHzatmega168   ATmega168 (3.3V, 8 MHz)
pro        8MHzatmega328   ATmega328P (3.3V, 8 MHz)
kevin@kevin-asrock:~/Arduino/Blink$
```

The requiqred files are there:
```
kevin@kevin-asrock:~/Arduino/Blink$ ls -ld /home/kevin/Arduino/arduino-1.8.12
drwxr-xr-x 10 kevin kevin 4096 Feb 13 04:32 /home/kevin/Arduino/arduino-1.8.12
kevin@kevin-asrock:~/Arduino/Blink$ ls -la /etc/avrdude.conf
-rw-r--r-- 1 root root 485369 Aug  6  2017 /etc/avrdude.conf
kevin@kevin-asrock:~/Arduino/Blink$ ls -la /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk
-rw-r--r-- 1 kevin kevin 69936 Mar 21 10:18 /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk
kevin@kevin-asrock:~/Arduino/Blink$
```

And yet, I get:
```
kevin@kevin-asrock:~/Arduino/Blink$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/kevin/Arduino/arduino-1.8.12 
- [USER]               ARDMK_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 1812 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/kevin/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/kevin/Arduino (from arduino preferences file)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [USER]               AVR_TOOLS_DIR = /usr 
- [COMPUTED]           ARDUINO_LIB_PATH = /home/kevin/Arduino/arduino-1.8.12/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/kevin/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 16MHzatmega328 
- [USER]               BOARD_TAG = pro 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-pro-16MHzatmega328 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 
- [COMPUTED]           DEVICE_PATH = /dev/ttyUSB1 (from MONITOR_PORT)
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 5.4.0 (avr-gcc)
-------------------------
mkdir -p build-pro-16MHzatmega328
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu= -DF_CPU= -DARDUINO=1812 -DARDUINO_ARCH_AVR -I/home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/cores/arduino -I/home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/cores/arduino/api -I/home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/variants/    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always Blink.ino -o build-pro-16MHzatmega328/Blink.ino.o
avr-g++: error: missing device or architecture after ‘-mmcu=’
/home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk:1421: recipe for target 'build-pro-16MHzatmega328/Blink.ino.o' failed
make: *** [build-pro-16MHzatmega328/Blink.ino.o] Error 1
kevin@kevin-asrock:~/Arduino/Blink$
```

Can anyone offer and suggestions or advice as to what I'm doing wrong?

Thanks so much.

-Kevin

sej7278, 2020-06-04:
this works for me:

```make
ARDUINO_DIR = /home/user/arduino-1.8.12
BOARD_TAG = pro
BOARD_SUB = 16MHzatmega328
include /home/user/Arduino-Makefile/Arduino.mk
```

kzembower, 2020-06-05:
Yeah, I can't tell why a similar file (modified for my environment) is not working for me:
```
kevin@kevin-asrock:~/Arduino/Blink$ cat Makefile
ARDUINO_DIR = /home/kevin/Arduino/arduino-1.8.12
BOARD_TAG = pro
BOARD_SUB = 16MHzatmega328
include /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk
kevin@kevin-asrock:~/Arduino/Blink$ ll /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk 
-rw-r--r-- 1 kevin kevin 69936 Mar 21 10:18 /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk
kevin@kevin-asrock:~/Arduino/Blink$ ls -ld /home/kevin/Arduino/arduino-1.8.12
drwxr-xr-x 10 kevin kevin 4096 Feb 13 04:32 /home/kevin/Arduino/arduino-1.8.12
kevin@kevin-asrock:~/Arduino/Blink$ make
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/kevin/Arduino/arduino-1.8.12 
- [USER]               ARDMK_DIR = /usr/share/arduino 
- [AUTODETECTED]       ARDUINO_VERSION = 1812 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/kevin/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/kevin/Arduino (from arduino preferences file)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [BUNDLED]            AVR_TOOLS_DIR = /home/kevin/Arduino/arduino-1.8.12/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /home/kevin/Arduino/arduino-1.8.12/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/kevin/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 16MHzatmega328 
- [USER]               BOARD_TAG = pro 
- [COMPUTED]           CORE =  (from build.core)
- [COMPUTED]           VARIANT =  (from build.variant)
- [COMPUTED]           OBJDIR = build-pro-16MHzatmega328 (from BOARD_TAG)
- [DEFAULT]            ARDUINO_CORE_PATH = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/cores/arduino 
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 
- [AUTODETECTED]       DEVICE_PATH = /dev/ttyUSB0 
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 7.3.0 (avr-gcc)
-------------------------
mkdir -p build-pro-16MHzatmega328
/home/kevin/Arduino/arduino-1.8.12/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu= -DF_CPU= -DARDUINO=1812 -DARDUINO_ARCH_AVR -I/home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/cores/arduino -I/home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/cores/arduino/api -I/home/kevin/Arduino/arduino-1.8.12/hardware/arduino/avr/variants/    -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always Blink.ino -o build-pro-16MHzatmega328/Blink.ino.o
avr-g++: error: missing device or architecture after '-mmcu='
/home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk:1421: recipe for target 'build-pro-16MHzatmega328/Blink.ino.o' failed
make: *** [build-pro-16MHzatmega328/Blink.ino.o] Error 1
kevin@kevin-asrock:~/Arduino/Blink$ 
```
Let me know if you have any thoughts or suggestions:
-Kevin

sej7278, 2020-06-05:
can you try my makefile with a simple blink.ino or something, as you're not getting CORE/VARIANT amongst others, my output is:

```
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/user/arduino-1.8.12 
- [COMPUTED]           ARDMK_DIR = /home/user/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 1812 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/user/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/user/arduino16 (from arduino preferences file)
- [COMPUTED]           ARDUINO_VAR_PATH = /home/user/arduino-1.8.12/hardware/arduino/avr/variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /home/user/arduino-1.8.12/hardware/arduino/avr/boards.txt (from ARDUINO_DIR)
- [BUNDLED]            AVR_TOOLS_DIR = /home/user/arduino-1.8.12/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /home/user/arduino-1.8.12/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/user/arduino-1.8.12/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/user/arduino16/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_SUB = 16MHzatmega328 
- [USER]               BOARD_TAG = pro 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = eightanaloginputs (from build.variant)
- [COMPUTED]           OBJDIR = build-pro-16MHzatmega328 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/user/arduino-1.8.12/hardware/arduino/avr/cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
- [COMPUTED]           BOOTLOADER_PARENT = /home/user/arduino-1.8.12/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 7.3.0 (avr-gcc)
```

kzembower, 2020-06-06:
Yes, as I said in my previous message, I AM using your makefile, with the only changes required to go from your environment to mine:
```
kevin@kevin-asrock:~/Arduino/Blink$ cat Makefile
ARDUINO_DIR = /home/kevin/Arduino/arduino-1.8.12
BOARD_TAG = pro
BOARD_SUB = 16MHzatmega328
include /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk
kevin@kevin-asrock:~/Arduino/Blink$ 
```
I am trying to compile the standard example Blink.ino from the Arduino distribution. It is:
```
kevin@kevin-asrock:~/Arduino/Blink$ cat Blink.ino 
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
kevin@kevin-asrock:~/Arduino/Blink$ 
```
I have both the Makefile and Blink.ino located in the directory ~/Arduino/Blink/, and I cd there and run 'make'.

This is why it's so confusing to me; I think I'm doing everything the same as you. In addition, it was working just fine for two days, then stopped working for an unknown reason.

Thanks for your thoughts and suggestions. Are the any diagnostics or verbose outputs I can run to  help troubleshoot the problem? What part of Arduino.mk populates the 'mmcu=' option?

Thanks, again.

-Kevin

kzembower, 2020-06-12:
I discovered the difference between working and not working.

This works:
```BOARD_TAG = pro
BOARD_SUB = 16MHzatmega328
ARDUINO_DIR = /home/kevin/Arduino/arduino-1.8.12
ARDMK_DIR = /usr/share/arduino
include /usr/share/arduino/Arduino.mk
```
This uses the Arduino.mk file at /usr/share/arduino/Arduino.mk, which is version 1.5.2.

This DOES NOT WORK:
```BOARD_TAG = pro
BOARD_SUB = 16MHzatmega328
ARDUINO_DIR = /home/kevin/Arduino/arduino-1.8.12
ARDMK_DIR = /usr/share/arduino
# include /usr/share/arduino/Arduino.mk
include /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk
```
This uses the Arduino.mk file at /home/kevin/Arduino/Arduino-Makefile-master/Arduino.mk, which is version 1.6.0.

Hope this is helpful for troubleshooting.

-Kevin

closed 2020-06-12 by kzembower