sudar/Arduino-Makefile #642: List of option

horchi, 2020-08-28:
very nice project! Much more comfortable to use Makefiles instead of the IDE :) 

One question, where I find a list of all needed options?

I had to set AVRDUDE_ARD_BAUDRATE to get the upload running, first I tried to set the MONITOR_BAUDRATE but it had no effect to the call of  avrdude.

The automatic detected Rate also not worked. By setting: AVRDUDE_ARD_BAUDRATE = 57600 all is fine now.

Thx and best regards Jörg

sej7278, 2020-08-28:
the [blink](https://github.com/sudar/Arduino-Makefile/tree/master/examples/Blink) example is useful, the options are all documented [here](https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md)

horchi, 2020-08-29:
Many thanks, [https://github.com/sudar/Arduino-Makefile/blob/master/arduino-mk-vars.md](this) is exactly hat I'am looking for.
May It make sense to put this link also into the main README.md?

sej7278, 2020-08-29:
There is already a link from readme

https://github.com/sudar/Arduino-Makefile#useful-variables

horchi, 2020-08-30:
thx! Sorry I had totally overseen this

closed 2020-08-30 by horchi