sudar/Arduino-Makefile #644: ELF link uses $(CC), not $(CXX)

kpfleming, 2020-10-18:
I've just started using this Makefile kit (it's quite nice) for a SAMD-based small project: the project currently contains one .ino file and some .hpp files (which are '#include'-d into the .ino file for the moment).

I got everything to compile, but the final link to produce the ELF file failed with a number of missing symbols, all of which come from the C++ standard library. While looking at Arduino.mk [here](https://github.com/sudar/Arduino-Makefile/blob/c7eb909b849e354fdb7630d11405ea6de706f1ba/Arduino.mk#L1679) I noticed that the linking is being done by calling $(CC) instead of $(CXX), which means the C++ standard library doesn't get included into the link automatically.

While I've made this change in my local Arduino.mk and now the project links successfully, I can't help but think that this is something which shouldn't be necessary, as pretty much all Arduino projects (at least those which originated in the IDE) are in C++ and require the standard library. Is there some sort of configuration setting I've missed?

tuna-f1sh, 2020-10-19:
Yes looks like you are right. The Arduino IDE also calls `g++` for linking this is an oversight in the current SAM support. I think it hasn't been picked up prior to this since I've not used the C++ standard library with Arduino and most libraries don't. 

Suggest you make a pull request with this minor change. Thanks.

sej7278, 2020-10-20:
merged thanks

PR #645 (2020-10-20): Use C++ compiler when linking for SAMD-based boards
The Arduino IDE supports full usage of C++ (including the standard
library) on SAMD-based boards, so the final linking of the executable
must be done using the C++ compiler.

Closes #644.
commit 37e75759 (2020-10-20): Use C++ compiler when linking for SAMD-based boards

The Arduino IDE supports full usage of C++ (including the standard
library) on SAMD-based boards, so the final linking of the executable
must be done using the C++ compiler.

Closes #644.
closed 2020-10-20 by sej7278