sudar/Arduino-Makefile #649: plugin needed to handle lto object

cdupont, 2020-12-14:
I ma trying to compile with Arduino MKR NB 1500.
Here is my Makefile:
```
ARDUINO_DIR      = /home/cdupont/Documents/workspace/arduino-1.8.10/
BOARD_TAG        = mkrnb1500
USER_LIB_PATH    = /home/cdupont/Documents/Waziup/WaziDev/libraries
MONITOR_PORT     = /dev/ttyUSB0
ARDUINO_LIBS     = PString-Arduino-lib MKRNB ArduinoHttpClient
ARDUINO_PACKAGE_DIR = /home/cdupont/.arduino15/packages/

include /home/cdupont/Documents/workspace/Arduino-Makefile/Sam.mk
```

I get the following errors:
```
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar: build-mkrnb1500/userlibs/MKRNB/src/NB_SMS.cpp.o: plugin needed to handle lto object
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar: build-mkrnb1500/userlibs/MKRNB/src/GPRS.cpp.o: plugin needed to handle lto object
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar: build-mkrnb1500/userlibs/MKRNB/src/NBPIN.cpp.o: plugin needed to handle lto object
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar: build-mkrnb1500/userlibs/MKRNB/src/NBScanner.cpp.o: plugin needed to handle lto object
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar: build-mkrnb1500/userlibs/MKRNB/src/Modem.cpp.o: plugin needed to handle lto object
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar: build-mkrnb1500/userlibs/PString-Arduino-lib/PString.cpp.o: plugin needed to handle lto object
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-gcc -T/home/cdupont/.arduino15/packages//arduino/hardware/samd/1.8.9/variants/mkrnb1500/linker_scripts/gcc/flash_with_bootloader.ld -Wl,-Map=build-mkrnb1500/compilation.map -o build-mkrnb1500/compilation.elf build-mkrnb1500/peetBrosLib.cpp.o build-mkrnb1500/compilation.ino.o   --specs=nano.specs --specs=nosys.specs -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--start-group -larm_cortexM0l_math -L/home/cdupont/.arduino15/packages//arduino/tools/CMSIS/4.5.0/CMSIS/Lib/GCC -lm -mcpu=cortex-m0plus -Wl,--gc-sections -Os -flto -fuse-linker-plugin build-mkrnb1500/libcore.a -Wl,--end-group
/home/cdupont/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: warning: cannot find entry symbol Reset_Handler; defaulting
to 0000000000002000
/tmp/ccFY58ZT.ltrans0.ltrans.o: In function `NB::~NB()':
<artificial>:(.text+0x6): undefined reference to `String::~String()'
/tmp/ccFY58ZT.ltrans0.ltrans.o: In function `_GLOBAL__sub_I_client':
<artificial>:(.text.startup+0x14): undefined reference to `IPAddress::IPAddress(unsigned char, unsigned char, unsigned char, unsigned char)'
<artificial>:(.text.startup+0x1c): undefined reference to `NBClient::NBClient(bool)'
<artificial>:(.text.startup+0x32): undefined reference to `GPRS::GPRS()'
<artificial>:(.text.startup+0x46): undefined reference to `NB::NB(bool)'
<artificial>:(.text.startup+0x60): undefined reference to `NBClient::~NBClient()'
<artificial>:(.text.startup+0x64): undefined reference to `GPRS::~GPRS()'
collect2: error: ld returned 1 exit status
```

Thanks for the help!

cdupont, 2020-12-15:
I made some progress by adding `--plugin liblto_plugin.so` on the ar commands:
```
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar rcs build-mkrnb1500/libcore.a   (...)  --plugin /home/cdupont/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/lib/gcc/arm-none-eabi/7.2.1/liblto_plugin.s
```
This way `ar` runs without complaining.

For the last step, I have to change `arm-none-eabi-gcc` with `arm-none-eabi-g++`:
```
/home/cdupont/.arduino15/packages//arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++  (...)
```

tuna-f1sh, 2020-12-16:
Are you using the latest commit in this repository? `gcc` was swapped for `g++` as the linker in this PR https://github.com/sudar/Arduino-Makefile/pull/645

tuna-f1sh, 2020-12-16:
Also, are you able to compile the ZeroBlink example? itoa.c is compiled as part of the SAMD core. I just re-tested and verified this:

`packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-ar rcs build-arduino_zero_native/libcore.a  build-arduino_zero_native/core/variant.cpp.o  build-arduino_zero_native/core/WInterrupts.c.o  build-arduino_zero_native/core/cortex_handlers.c.o  build-arduino_zero_native/core/delay.c.o  build-arduino_zero_native/core/hooks.c.o  build-arduino_zero_native/core/itoa.c.o  build-arduino_zero_native/core/pulse.c.o  build-arduino_zero_native/core/startup.c.o  build-arduino_zero_native/core/wiring.c.o  build-arduino_zero_native/core/wiring_analog.c.o  build-arduino_zero_native/core/wiring_digital.c.o  build-arduino_zero_native/core/wiring_private.c.o  build-arduino_zero_native/core/wiring_shift.c.o  build-arduino_zero_native/core/avr/dtostrf.c.o  build-arduino_zero_native/core/USB/samd21_host.c.o  build-arduino_zero_native/core/IPAddress.cpp.o  build-arduino_zero_native/core/Print.cpp.o  build-arduino_zero_native
/core/Reset.cpp.o  build-arduino_zero_native/core/SERCOM.cpp.o  build-arduino_zero_native/core/Stream.cpp.o  build-arduino_zero_native/core/Tone.cpp.o  build-arduino_zero_native/core/Uart.cpp.o  build-arduino_zero_native/core/WMath.cpp.o  build-arduino_zero_native/core/WString.cpp.o  build-arduino_zero_native/core/abi.cpp.o  build-arduino_zero_native/core/main.cpp.o  build-arduino_zero_native/core/new.cpp.o  build-arduino_zero_native/core/USB/CDC.cpp.o  build-arduino_zero_native/core/USB/PluggableUSB.cpp.o  build-arduino_zero_native/core/USB/USBCore.cpp.o  build-arduino_zero_native/core/pulse_asm.S.o `

I've also confirmed the IDE doesn't add this flag so I'm not sure what is going on here.

cdupont, 2020-12-16:
Yes, that's great, using the latest source of this repository solved the problem.
Let me close this issue.

closed 2020-12-16 by cdupont