sudar/Arduino-Makefile #652: monitor: why not use "cat"? (Question)

cdupont, 2020-12-28:
I see Arduino-Makefile uses "screen" for the monitor. However, it's not very well know for many users...
Why not using a simple "cat"? I use this:

monitor:
         stty -F $(MONITOR_PORT) raw $(MONITOR_BAUDRATE)
         cat $(MONITOR_PORT)

It works well. You can quit with Ctrl-C.

sej7278, 2020-12-28:
as i recall cat has buffering issues when you change baudrate - you see the previous output when you make the baudrate faster or something?

also stty is probably more unknown than screen and has differences between macos and linux versions.

you can change MONITOR_CMD in your local Makefile or bind ctrl-c to the kill command in gnu-screen

rohoog, 2021-04-02:
I would say that a more important reason is that cat is uni-directional: you cannot type any text to your arduino serial port and expect it to be readable using Serial.read or Serial.readBytes, while with gnu-screen you can.