sudar/Arduino-Makefile #658: Some flags are not parsed from the boards.txt for SAMD

JDuchniewicz, 2021-04-21:
I am

JDuchniewicz, 2021-04-21:
Okay, the issue was with supplying only `nano` instead of `nano_33_iot` because that is how the Bare-Arduino-Makefile instructed. So it looked like this:
```
BOARD_TAG = nano
BOARD_SUB = nano_33_iot
``` 
instead of
```
BOARD_TAG = nano_33_iot
```

closed 2021-04-21 by JDuchniewicz

tuna-f1sh, 2021-04-22:
Just for reference in case you come across this again, `nano` in that project will be referring to the original AVR Arduino Nano https://www.arduino.cc/en/pmwiki.php?n=Main/ArduinoBoardNano

The Nano IoT 33 is a much newer board so you can assume any documentation with `nano` is the AVR board.

JDuchniewicz, 2021-04-22:
Thanks! I uploaded the [PR](https://github.com/ladislas/Bare-Arduino-Project/pull/66/) to the other repo with clarification for the SAM boards.