sudar/Arduino-Makefile #668: Arduino.h no such file or directory

UltraBlackLinux, 2021-08-04:
Hey there,
I've been trying to get this to work for quite some time now, but It's just not working out.
I have installed arduino, and even redownloaded the arduino-core to a custom directory, but still Ar-Mf is not detecting the Arduino.h file, and I am now out of Ideas.
What am I doing wrong?

This is the terminal output: 
```
mkdir -p build-uno
/usr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=100 -DARDUINO_AVR_UNO  "-DARDUINO_BOARD=\"AVR_UNO\"" "-DARDUINO_VARIANT=\"standard\"" -I/home/maus/Arduino/.ArduinoCore-avr//hardware/arduino//cores/arduino -I/home/maus/Arduino/.ArduinoCore-avr//hardware/arduino//cores/arduino/api -I/home/maus/Arduino/.ArduinoCore-avr//hardware/arduino//variants/standard    -Wall-ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always flash.ino -o build-uno/flash.ino.o
<command-line>: fatal error: Arduino.h: No such file or directory
compilation terminated.
make: *** [/usr/share/arduino/Arduino.mk:1446: build-uno/flash.ino.o] Error 1
```

Thanks!

tuna-f1sh, 2021-08-05:
We need the full header that is printed at the start to really debug anything but first thing I would ask is does '/home/maus/Arduino/.ArduinoCore-avr//hardware/arduino//cores/arduino' actually exist and contain the Arduino core? I think there is a missing ARCHITECTURE variable missing the that path. 'arduino//cores' should be 'arduino/avr/cores'.

UltraBlackLinux, 2021-08-05:
> We need the full header that is printed at the start to really debug anything but first thing I would ask is does '/home/maus/Arduino/.ArduinoCore-avr//hardware/arduino//cores/arduino' actually exist and contain the Arduino core? I think there is a missing ARCHITECTURE variable missing the that path. 'arduino//cores' should be 'arduino/avr/cores'.
> 
> Can you try compiling the 'Blink' example (with the core set to the custom directory you downloaded) too?

Oh well, the directory actually doesn't exist, so I set it back to what it was before (/usr/share/arduino), but it still doesn't work - I only have the directories up to .../avr (The boards.txt didn't generate for me so I had to manually create the directories - What a buggy mess - so this might have broken something)

UltraBlackLinux, 2021-08-05:
https://github.com/sudar/Arduino-Makefile/issues/466
I had to install avr-libc :facepalm:
Why is this such a mess XD

UltraBlackLinux, 2021-08-05:
Huh? Do I really need to manually add every library? This can't be right - It's also missing built in data types?:
[image]

UltraBlackLinux, 2021-08-05:
Welp this is weird, I have to include arduino.h manually everytime

UltraBlackLinux, 2021-08-05:
Aaa this is driving me crazy. I am having more and more library problems and they all seem to disappear, when new come.
Now AR_MK can't find the wire library, even though I installed it for sure

UltraBlackLinux, 2021-08-05:
I wonder, how often I will reopen this...
Please excuse this.

closed 2021-08-05 by UltraBlackLinux